robotology/mesh-support #1579: iRonCub3: smoke from a board on the back

gabrielenava, 2023-06-15:
### Device name 🤖

iCubGenova09 - iRonCub3

### Request/Failure description

There was smoke and smell of something burning from the upper back of the robot, close to the battery display.

### Detailed context

We connected the robot to the power supply and turned on motors and on board pc. We did not start the yarprobotinterface. After few minutes, we smelled something burning. We saw smoke coming out from the robot upper back (close to the battery display) and we immediately switched off the power.

### Additional context

Blocking iRonCub3 experiments and demos cc @DanielePucci

sgiraz, 2023-06-16:
Hi @gabrielenava,

In accordance with @maggia80, we are going to check the status of the robot, in particular the battery and the related electronics with @MrAndrea.

cc @AntonioAzocar @Gandoo @Fabrizio69

MrAndrea, 2023-06-16:
I checked the robot together with @Gandoo.
The problem was with the 5V regulator on the BAT board, it provides... 20V instead of 5V. The smoke comes from the unfortunate display on the back, which is the only device connected to that regulator.
I don't know the cause of this, maybe a dropped screw or debris that got inside the covers?
There are no other problems.

Therefore, the solution is to replace the BAT board (iitcode 10050.D) and the display (iitcode 7282) with the microSD (iitcode 15022).

Note:
- before mounting the new BAT, I have to check if the firmware revision is the 1.2.0 (see [this PR](https://github.com/robotology/icub-firmware-build/pull/86))
- the new display must be programmed

gabrielenava, 2023-06-16:
thank you @MrAndrea!

MrAndrea, 2023-06-19:
I programmed a new BAT board and the display. We can proceed with the replacement.

cc @Gandoo @Fabrizio69

MrAndrea, 2023-06-19:
Together with @Gandoo we replaced the faulty boards. 
Power tests were done (both battery and power supply) and were successful, the robot now works again.

cc @Fabrizio69 @maggia80

gabrielenava, 2023-06-20:
today afternoon me and @vpunithreddy tried to start the robot, bu there were issues: see https://github.com/robotology/icub-tech-support/issues/1581

I don't think the problem is correlated, but it needs to be fixed in order to complete the testing. 

Anyways we also connected the robot to the power supply and there was no smoke or burning smell.

A question for @MrAndrea: we are using these settings with our power supply:

[image]

we never changed them since we got the power supply, are they the correct ones for iCub3?

MrAndrea, 2023-06-21:
Yes @gabrielenava, the settings of the power supply in the [above comment](https://github.com/robotology/icub-tech-support/issues/1579#issuecomment-1599208016) are correct.

| Voltage | 44V |
|:-:|:-:|
| Current | 25A |
| Power | 1500W|

AntonioConsilvio, 2023-06-21:
Thanks @MrAndrea and @Gandoo, great job! 🚀 
We will organise to solve the other problem as well ([#1581](https://github.com/robotology/icub-tech-support/issues/1581))

Cc @Fabrizio69

closed 2023-06-21 by AntonioConsilvio