robotology/mesh-support #1580: ergoCubSN000 right thumb OC gets stuck

S-Dafarra, 2023-06-16:
### Device name 🤖

ergoCub S/N000

### Request/Failure description

We noticed that at a certain point, the right thumb stopped following commands.

### Detailed context

The joint appears to be in position mode, but it does not follow the commands and the motor position is stuck to a given value.

sgiraz, 2023-06-16:
Hi @AntonioConsilvio,

May the issue reported here correlated to the #1578 ?

AntonioConsilvio, 2023-06-22:
Hi @sgiraz the problem is unrelated. 

However, the problem was that the MotorPosition of the Thumb OpenClose reached the value of [rotoPosMax](https://github.com/robotology/robots-configuration/blob/d106aacf66104e14fc93a14a3e17e0311fdd6b12/ergoCubSN000/hardware/mechanicals/right_arm-eb22-j7_10-mec.xml#L30).

During the demo we changed this value from -6500 to -100, but the robot needs someone to hold the finger open during calibration to calibrate correctly.

This problem is probably due to too long a tendon, so we will organise an intervention to solve it.

cc @maggia80 @S-Dafarra

S-Dafarra, 2023-06-30:
cc @marcellosavoldi

sgiraz, 2023-07-26:
A PRs is coming to fix this issue:
- https://github.com/robotology/icub-firmware-build/pull/95

AntonioConsilvio, 2023-08-01:
Hi @S-Dafarra @MSECode, me and @fbiggi tightened the thumb tendon open close and it seems to behave better.

https://github.com/robotology/icub-tech-support/assets/114915464/8125c8bf-fcf9-41fd-9779-b162aaf77287

The abundance of the tendon has been eliminated as much as possible.

cc @sgiraz

closed 2023-08-02 by AntonioConsilvio