robotology/mesh-support #1581: iRonCub3: timeout on right arm board EB3

gabrielenava, 2023-06-20:
### Device name 🤖

iCubGenova09 - iRonCub3

### Request/Failure description

We tried to start the robot to verify the status after #1579 

the `yarprobotinterface` fails with a timeout error on several boards, the first one that fails is the EMS `EB3` from the right arm.

### Detailed context

Running `Firmwareupdater` confirmed that the only visible boards are the head and the left arm. I suspect there is a problem with the EMS on the right arm and then all the other boards in the chain are not seen because of that.

[image]

gabrielenava, 2023-06-21:
title and first comment edited as EB3 is not on the right arm but on the chest.

Gandoo, 2023-06-21:
ciao,  after a check because it worked at times there was the E26 cable connector on the EB3 board that lost the signal. I replaced crimp and connector, now it works

cc @gabrielenava @vpunithreddy @AntonioConsilvio @sgiraz @DanielePucci

gabrielenava, 2023-06-21:
thank you! today I worked with the robot and everything was fine!

AntonioConsilvio, 2023-06-22:
Good job guys! Closing ✔️

cc @gabrielenava @Gandoo @Fabrizio69

closed 2023-06-22 by AntonioConsilvio