robotology/mesh-support #1582: iCubGenova11 - Right thumb tendons broken

xEnVrE, 2023-06-21:
### Device name 🤖

iCubGenova11

### Request/Failure description

While doing the demo for the FamilyDay, one of the tendons of the right thumb broke.

### Detailed context

At the moment we disabled the right arm to let the demos continue.

### Additional context

cc @mfussi66

GiacomoVassallo13, 2023-06-23:
Hi. The proximal tendon of thumb has been replaced. Also replaced the sheaths that were damaged. . In addition I proceeded to stretch the proximal tendons of the middle and index finger, as well as the thumb oppose. 
While I was making the repairs I noticed that some screws on the fingers were loose so I proceeded to tighten them

cc: @Fabrizio69 @AntonioConsilvio @sgiraz

xEnVrE, 2023-06-23:
Thanks @GiacomoVassallo13

xEnVrE, 2023-07-03:
This week I will use the robot and give feedback on this.

sgiraz, 2023-07-04:
I'm going to close this issue. Feel free to reopen if needed.

closed 2023-07-04 by sgiraz