robotology/mesh-support #1583: [iCubErzelli02] Right shoulder tendon broken (right arm)

ddetommaso, 2023-06-22:
### Device name 🤖

iCubErzelli02

### Request/Failure description

one of the tendon broke and is out of its pulley

### Detailed context

-

### Additional context

[image]

### How does it affect you?

Data collection is done. Thus, we have no urgency to fix the robot and we would ask you to intervene possibly on the days Thu 29th/Fri 30th.

AntonioConsilvio, 2023-06-27:
Hi @ddetommaso, We agree to intervene on Friday 30th

cc @Fabrizio69

AntonioConsilvio, 2023-07-03:
Hi @ddetommaso, me and @GiacomoVassallo13 replaced the tendon and tested the robot on Friday 30th.

The shoulder of the robot works properly now.

cc @Fabrizio69

closed 2023-07-03 by AntonioConsilvio