robotology/mesh-support #1584: `ergoCubSN000` right knee goes into hardware fault i2t

mebbaid, 2023-06-22:
### Device name 🤖

ergoCubSN000

### Request/Failure description

`r_knee` joint moves very fast and in a strange manner for a small change in position reference when controlled in position from the `yarpmotorgui`.

The fault indicates an i2t fault. Attached is a video of the issue.

### Detailed context

https://github.com/robotology/icub-tech-support/assets/36491318/00c5e422-1be5-4f27-8b0c-4f8945a6aa1f

`r_knee` joint moves very fast and in a strange manner for a small change in position reference when controlled in position from the `yarpmotorgui`.

The fault indicates an i2t fault. Attached is a video of the issue.

### Additional context

We were trying to test the walking.

mebbaid, 2023-06-22:
I restarted the robot motors, and the issue was not reproduced. I am not sure about the source of the problem (perhaps encoders or calibration or both). In case you want to close this issue feel free, or we can leave it open in case it happens in the next few days.

AntonioConsilvio, 2023-06-27:
Hi @mebbaid, as the problem has not reoccurred I am closing the issue, feel free to reopen it if the problem occurs again.

closed 2023-06-27 by AntonioConsilvio