robotology/mesh-support #1585: iRonCub3: the robot faults itself when touching the back cover

gabrielenava, 2023-06-22:
### Device name 🤖

iCubGenova09 - iRonCub3

### Request/Failure description

If someone touches the back cover, close to the power supply connector, the robot joints go in hardware fault or idle.

We noted that the fault support is broken: may be related to this issue?

### Detailed context

https://github.com/robotology/icub-tech-support/assets/12396934/715050a6-cf69-4027-b614-106f6e271cf1

### How does it affect you?

cc @DanielePucci @CarlottaSartore @lrapetti @HosameldinMohamed @vpunith

maggia80, 2023-06-23:
tech-support I would check the FED board

AntonioConsilvio, 2023-06-27:
Hi @gabrielenava, I replaced the holder of the FED board because it was damaged: 

[image]

The FED board is in very good condition so I did not replace it.  

However, I realised that the problem was the wireless relay, because when someone touched it the robot would fall in idle. 

So probably the internal circuit of the relay is damaged.

So, I removed the relay and closed the fault circuit and now the robot only works with the red button, but it was tested and now the problem does not happen again.

cc @maggia80 @Fabrizio69

sgiraz, 2023-07-04:
> So probably the internal circuit of the relay is damaged.
> 
> So, I removed the relay and closed the fault circuit and now the robot only works with the red button, but it was tested and now the problem does not happen again.

Thanks @AntonioConsilvio!
As of now, we have to (order?) and mount a new relay to restore the wireless fault functionality.
@Fabrizio69 @gabrielenava, please talk to each other to decide who should provide the new relay.

gabrielenava, 2023-07-25:
>@Fabrizio69 @gabrielenava, please talk to each other to decide who should provide the new relay.

It seems we purchased the relay last time, I can have a look at the old issue (cc @S-Dafarra) and open a new purchase

S-Dafarra, 2023-07-26:
> > @Fabrizio69 @gabrielenava, please talk to each other to decide who should provide the new relay.
> 
> It seems we purchased the relay last time, I can have a look at the old issue (cc @S-Dafarra) and open a new purchase

Back in time I bought it myself and then asked for reimbursement. It was on Amazon and I needed it quickly

AntonioConsilvio, 2023-08-22:
> So, I removed the relay and closed the fault circuit and now the robot only works with the red button, but it was tested and now the problem does not happen again.

After my intervention the robot does not fault itself anymore. So I consider the issue solved. 

If you need further support to install a new wireless relay, I would ask you to open a new ticket for this task. 

Closing.

cc @sgiraz @gabrielenava

closed 2023-08-22 by AntonioConsilvio