robotology/mesh-support #1587: [ergoCubSN000] ergoCub power goes to 0VP

lrapetti, 2023-06-26:
### Device name 🤖

ergoCubSN000

### Request/Failure description

The power supply goes in protection when we move the power cable.

### Detailed context

It happens only if the Sense is connected, disconnecting the sense it never happens.

https://github.com/robotology/icub-tech-support/assets/35487806/af96311b-0bf5-46cc-a0ae-613bdbde8464

cc @maggia80 @andre

AntonioConsilvio, 2023-06-27:
Hi, the power supply shows this error: 

[image]

It it happens when someone touch the Sense cable. 

cc @maggia80 @MrAndrea

maggia80, 2023-06-27:
ok, make sense. We should check if the remote sense is well connected.

sgiraz, 2023-07-12:
Hi @AntonioAzocar, 

The robot is now operative again. We can proceed to test the sense with the robot attached.

sgiraz, 2023-07-14:
Ok, we spotted that the red (positive) cable wasn't properly connected to the power supply `sense` connector.
Once reconnected correctly the problem is solved.

Thanks, @AntonioAzocar and @Gandoo for the check and fix.
Closing.

@lrapetti

closed 2023-07-14 by sgiraz