robotology/mesh-support #1588: Right knee motor problem

gabrielenava, 2023-06-28:
### Device name 🤖

iCubGenova09 - iRonCub3

### Request/Failure description

When moving in position control, the right leg knee joint vibrates and makes noise without actually moving.

The encoders readings from the yarpmotorgui seem fine. If it does not move, the joint does not shake.

I'll add the logger below:

### Detailed context

https://github.com/robotology/icub-tech-support/assets/12396934/a8167622-2547-4263-a441-76db0429151c

cc @vpunithreddy @DanielePucci

sgiraz, 2023-07-05:
Hi @gabrielenava,

It may be necessary to clean the optical disc or open the motor for further inspection.

cc @Fabrizio69 @fbiggi

sgiraz, 2023-07-06:
Hi @gabrielenava, may you drop here the log, please?

gabrielenava, 2023-07-06:
there were some apparently uncorrelated errors the first time the problem occurred, this is why in the above comment I wrote that I'll upload also the logger. But during the last experimental campaign the problem occurred again multiple times and there were no errors anymore in the logger, so we did not save it.

>It may be necessary to clean the optical disc or open the motor for further inspection.

the robot is already downstairs, please tell me if you want  me to bring it in the icub-tech support room

Fabrizio69, 2023-07-06:
Hi @gabrielenava, does this problem block your activities? I ask to plan this activity...

gabrielenava, 2023-07-06:
>Hi @gabrielenava, does this problem block your activities? I ask to plan this activity...

unfortunately yes, because the problem occur at startup when the knee joint calibrates, and the robot shakes a lot because of that...it may trigger other failures. We tried to change the calibration type so that it does not move the knee, but then there were issues with the other leg joints and it did not seem to me so straightforward to do it

sgiraz, 2023-07-10:
Hi @gabrielenava, 

After the optical disc has been cleaned, the issue should be solved.
Let us know.

gabrielenava, 2023-07-11:
hello @sgiraz 

we started the robot many times after the intervention. Once, the knee did not move during the calibration, but it was an isolated case. all the other times it worked, so for now I think we can close the issue. If the problem will appear again in the future, I will open a new issue

sgiraz, 2023-07-11:
Ok, thanks @gabrielenava.
Closing.

closed 2023-07-11 by sgiraz