robotology/mesh-support #1589: ergoCubSN000 right adduction not moving

S-Dafarra, 2023-06-30:
### Device name 🤖

ergoCub S/N000

### Request/Failure description

While shooting the robot while opening a light handle, the right thumb adduction stopped moving and it was not possible to recover ita

### Detailed context

The encoders seem to measure correctly, but the joint seems to push toward the limit and gets stuck

sgiraz, 2023-07-18:
Hi @S-Dafarra,

This afternoon @AntonioConsilvio found an MMA's PCB fap connector of the right hand faulted. He replaced the MMA board with a new one. I don't remember if it was related to the adduction or OC, but surely it was a problem.
Please, let us know if you have the opportunity to try it. At any rate, we'll test it asap.

see also #1580

AntonioConsilvio, 2023-08-01:
Today I tried adduction of the right thumb and it does not push to the limit and does not get stuck: 

https://github.com/robotology/icub-tech-support/assets/114915464/5907a415-0365-4689-97bc-0eb78aeb155b

cc @sgiraz

closed 2023-08-02 by AntonioConsilvio