robotology/mesh-support #1590: iCubGenova11 - Left thumb distal tendon broken

xEnVrE, 2023-07-03:
### Device name 🤖

iCubGenova11

### Request/Failure description

As per the title, the left thumb distal tendon is broken.

### Detailed context

When the `yarprobotinterface` calibrates the robot, the noise of the motor spinning freely can be heard.

xEnVrE, 2023-07-03:
Possibly related to #1575

GiacomoVassallo13, 2023-07-14:
Hi. The thumb distal joint has been replaced. The tendom has been replaced and a finger tip cable has been re-soldered.
I attend feedback.

cc: @sgiraz

sgiraz, 2023-07-17:
Hi @xEnVrE, 

I guess this issue has been solved. In this case I proceed closing the issue.
Feel free to reopen if necesary.

closed 2023-07-17 by sgiraz

xEnVrE, 2023-07-17:
Thanks @sgiraz @GiacomoVassallo13