robotology/mesh-support #1591: iRonCub1: problem with right leg ankle board

gabrielenava, 2023-07-03:
### Device name 🤖

iRonCub1 - iCubGenova04

### Request/Failure description

The robot does not start, in the yarprobotinterface there is this error:

```
[ERROR]  from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 253s 946m 428u: (code 0x00000013, par16 0x0001 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
```

### Detailed context

we run firmwareupdater and we don't see the right foot FT sensor when we run `discover`.

cc @antonellopaolino

sgiraz, 2023-07-04:
Maybe we have to change the ems eb9 board because the power is 2.7V instead of 5V.\
We'll take a look at it asap.

gabrielenava, 2023-11-01:
keeping it open: the problem still persists. It would be nice to fix this and https://github.com/robotology/icub-tech-support/issues/1666 so that we can collect more data for the wind tunnel paper activity with @antonellopaolino

pattacini, 2023-11-01:
To prevent the automation from marking issues as https://github.com/robotology/icub-tech-support/labels/stale it suffices to use the label https://github.com/robotology/icub-tech-support/labels/pinned.

AntonioAzocar, 2024-09-19:
Hi!
Checking the output voltage from the CAN connector of the EMS4 to which the **9N2** cable is connected, we noted that it wasn't giving the required **5V**, but only **2** and obviously this voltage wasn't enough to power the FT sensor on the right foot.
So, in order to avoid changing the board directly, we took the power supply from the P8 connector and by doing this, the problem was solved and the robot works correctly.

[image]

[image]

cc @AntonioConsilvio

gabrielenava, 2024-09-23:
thank you @AntonioConsilvio and @AntonioAzocar! it seems working fine now!

AntonioConsilvio, 2024-09-24:
Well! Good job @AntonioAzocar. Closing! 🚀

closed 2024-09-24 by AntonioConsilvio