robotology/mesh-support #1592: [iCubErzelli01] Left arm thumb tendon broken

ddetommaso, 2023-07-04:
### Device name 🤖

iCubErzelli01

### Request/Failure description

While preparing a demo with grasping the left hand thumb broke.

### Detailed context

-

sgiraz, 2023-07-04:
Hi @ddetommaso, may you provide more information about this issue?
Is because of a broken tendon, or the motor faulted, or something else? 
Maybe some pictures may help to identify the possible problems.

ddetommaso, 2023-07-04:
> Hi @ddetommaso, may you provide more information about this issue?
> Is because of a broken tendon, or the motor faulted, or something else? 
> Maybe some pictures may help to identify the possible problems.

[image]

sgiraz, 2023-07-05:
Hi @ddetommaso,

The thumb joint of this robot has 3 different sub-joints: distal, proximal, and adduction
As we arrived on site, we spotted that the broken tendon was the adduction connected to motor M0.

[image]

Following the procedure described in the [documentation][1] (section 2.2), @GiacomoVassallo13  replaced the broken tendon with a new one.
After the replacement, we tested the joint using the Jogger and now it seems to work correctly:

https://github.com/robotology/icub-tech-support/assets/18039045/f5eb1106-c08e-46a0-8a84-3dbad86ebff4

⚠️ We'd need to perform further tests (e.g. using the yarprobotinterface), but I didn't know the password for the `iCub` account of the workstation next to the robot, hence I couldn't test it more deeply. 
However, next Monday (10th Jul) we planned to finalize the test and complete the maintenance activity also on `iCubErzelle02`  that today was under usage.

cc @Fabrizio69 

[1]: https://icub-tech-iit.github.io/documentation/icub_tendons/wrist_hand_cabling_V2/#21-location

ddetommaso, 2023-07-08:
Hi @sgiraz ,

thank you for the fix and the detailed explanation.
See you on Monday!

sgiraz, 2023-07-10:
### Maintenance activity report 10/07/23

**Interventions:**
| Description | pictures | Note |
| :---: | :---: | :---: |
| Replaced motor 1B3M1  |  | it was faulted |
| neck harmonic drive screws tightened correctly | | they were loosen |

**Tests:**
| Tests | pictures | Note |
| :---: | :---: | :---: |
| left arm joints | | |
| whole robot | | |
| demo to be performed next Wednesday | | |

**Notes:**
- 

cc @Fabrizio69

sgiraz, 2023-07-24:
Maintenance activity completed!
See also comment: https://github.com/robotology/icub-tech-support/issues/1570#issuecomment-1647587577

closed 2023-07-24 by sgiraz