robotology/mesh-support #1593: iCubGenova11 - Left thumb distal reads 90 degrees when fully open

xEnVrE, 2023-07-06:
### Device name 🤖

iCubGenova11

### Request/Failure description

When considering #1575, it was noticed that once the calibration is completed the left thumb distal is reading ~ 90 deegres when the joint is fully open - which is of course wrong.

### Detailed context

A picture showing the weird configuration:

[image]

AntonioConsilvio, 2023-07-18:
Hi @xEnVrE, yesterday I found the cable of one of the left thumb hall sensors damaged, and therefore to be replaced.

cc @sgiraz

xEnVrE, 2023-07-20:
Thanks, we can decide together the proper moment for fixing it.

xEnVrE, 2023-08-01:
Hi @sgiraz @AntonioConsilvio can I ask of there ie any update on the schedule of this activity?

Thank you

sgiraz, 2023-08-02:
Hi @xEnVrE,

As agreed, we scheduled the activity for tomorrow afternoon.

cc @AntonioConsilvio @GiacomoVassallo13

xEnVrE, 2023-08-03:
Thanks for the fix @AntonioConsilvio @Gandoo

AntonioConsilvio, 2023-08-04:
Hi @xEnVrE, I recalibrated and tested the distal joint of the left thumb. 

Now the joint is working properly. 

Remember to update the robots-configuration files:

- https://github.com/robotology/robots-configuration/pull/551

cc @sgiraz

closed 2023-08-04 by AntonioConsilvio