robotology/mesh-support #1595: [ergoCubSN000] Right elbow not moving smoothly

Woolfrey, 2023-07-06:
### Device name 🤖

ergoCubSN000

### Request/Failure description

The right elbow of the ergoCub is not moving smoothly when controlling it in position mode:

### Detailed context

When starting and calibrating the ergoCub robot today at 14:30, we noticed that the right elbow was not moving smoothly.

We ran the right arm motors in idle mode and we were able to move the joints by hand freely and without obstruction.

The technician from icub tech suspected it was a problem with the encoders on the elbow joint.

### Additional context

https://github.com/robotology/icub-tech-support/assets/62581255/09b2208c-cc4f-4adf-9467-46ddbb1feedb

### How does it affect you?

There is a visit to CRIS from Marco Giovannini and the ergoCub is needed to perform demonstrations. We need to do testing today to ensure everything is working for tomorrow.

sgiraz, 2023-07-07:
After the intervention of @Gandoo and @fbiggi, the right elbow would be restored to its normal functioning.
The issue was an LCORE broken wire.

Waiting for feedbacks

Woolfrey, 2023-07-07:
Yes, it was fixed shortly after I had posted this issue. It can be closed now.

closed 2023-07-11 by sgiraz