robotology/mesh-support #1596: iRonCub3 (iCubGenova09): torso pitch and yaw and left hip pitch do not start

gabrielenava, 2023-07-07:
### Device name 🤖

iRonCub3 - iCubGenova09

### Request/Failure description

we were testing the robot after #1594 

the torso pitch and yaw and the left hip pitch do not start during the calibration phase. On the yarpmotorgui, we see:

[image]

[image]

despite the message `hard limit reached` for the torso joints, in the `yarplogger` we can see that all joints show the `AEA encoder invalid data` message (together with the hard limits reached one):

[image]

### Detailed context

here the full logger:

[log_icub-head_yarprobotinterface[robotinterface]_1364.txt](https://github.com/robotology/icub-tech-support/files/11981059/log_icub-head_yarprobotinterface.robotinterface._1364.txt)

cc @STaliani @DanielePucci

gabrielenava, 2023-07-07:
cc @Gandoo

Gandoo, 2023-07-10:
[image]

cc @gabrielenava

sgiraz, 2023-07-11:
Hi @gabrielenava,

Has the problem been solved after @Gandoo's intervention?

gabrielenava, 2023-07-11:
hello @sgiraz 

the problem is now solved! thank you

sgiraz, 2023-07-11:
Nice! 🚀
Closing.

closed 2023-07-11 by sgiraz