robotology/mesh-support #1597: iRonCub3 (iCubGenova09): FT sensors frames in the arms and legs not correct (compared with ergoCubSN000

HosameldinMohamed, 2023-07-07:
### Device name 🤖

iCubGenova09

### Request/Failure description

Checking the Frames of the FT sensors in the arms and the hips for iRonCub3 (iCubGenova09) shows some differences from the hardware.

ergoCub FTs are mounted similarly but the URDF seems fine.

### Detailed context

Here's how the arms FTs are mounted in iRonCub3

<details>
<summary>Pics of arms FTs</summary>

[image]
[image]

[image]
[image]

</details>

Based on the documentation page https://icub-tech-iit.github.io/documentation/ft_sensors/ft_sensors/#physical-specifications the cable is in the direction of `-X`, but visualizing the frames of `iRonCub-Mk3` (located in [here](https://github.com/ami-iit/component_ironcub/tree/v4.1/models/iRonCub-Mk3/iRonCub/robots/iRonCub-Mk3)), the arms frames look like this:

<details>
<summary>iRonCub-Mk3 arms FTs frames</summary>

[image]

</details>

ergoCub-SN000 though seems fine (visualizing [this URDF](https://github.com/icub-tech-iit/ergocub-software/tree/v0.3.3/urdf/ergoCub/robots/ergoCubSN000))

<details>
<summary>ergoCub arms FTs frames</summary>

[image]

</details>

I "assume" the FT58 mounted in the legs are mounted the same way in `iRonCub-M3` (at the moment they are not mounted) as `ergoCubSN000`

<details>
<summary>ergoCub leg FT</summary>

[image]

</details>

But the frames are different

<details>
<summary>Frames of legs FTs</summary>

[image]

[image]

</details>

### Additional context

CC @DanielePucci

DanielePucci, 2023-07-07:
CC @pattacini @gabrielenava @GiulioRomualdi

HosameldinMohamed, 2023-07-07:
I was doing the above check with `iRonCub-Mk3` URDF assuming that its URDF is the same as `iCubGenova09`. Instead now I checked the latter and its **indeed correct**, only with different frame names (`l_arm_ft` instead of `l_arm_ft_sensor`)

URDF link (`v2.1.0`): https://github.com/robotology/icub-models/tree/v2.1.0/iCub/robots/iCubGenova09

[image]

So I guess this issue appeared during the generation of `iRonCub-Mk3` URDF from the assembly CC @vpunithreddy

HosameldinMohamed, 2023-07-07:
> So I guess this issue appeared during the generation of `iRonCub-Mk3` URDF from the assembly CC @vpunithreddy

It's probably related to URDF generation pipeline we are using for `iRonCub-Mk3` that results in some unexpected differences from the latest URDF of `iCubGenova09`. Manly in defining the FT sensors, for example, the `l_arm_ft_sensor` shows:

### In the current version of `iCubGenova09` URDF:

```xml
  <sensor name="l_arm_ft" type="force_torque">
    <parent joint="l_arm_ft_sensor"/>
    <force_torque>
      <frame>sensor</frame>
      <measure_direction>child_to_parent</measure_direction>
    </force_torque>
    <origin rpy="3.141592646862139 -3.10468332187885e-08 -1.5707962571980942" xyz="3.469446951953614e-18 0.0 0.0"/>
  </sensor>
```

### In the current version of `iRonCub-Mk3` URDF:

```xml
  <sensor name="l_arm_ft_sensor" type="force_torque">
    <parent joint="l_arm_ft_sensor"/>
    <force_torque>
      <frame>child</frame>
      <measure_direction>child_to_parent</measure_direction>
    </force_torque>
  </sensor>
```

So most likely the missing `origin` parameter is what caused the FT frames to be wrong in `iRonCub-Mk3`.

I will open another issue to check this out. But I will close this since no HW support is needed! Sorry for the confusion!

closed 2023-07-07 by HosameldinMohamed