robotology/mesh-support #1599: Support in the ARE application (software + hardware)

reafrancesco, 2023-07-13:
### Device name 🤖

iCubGenova07

### Request/Failure description

We tried the ARE demo on the iCubGenova07 and the iCub fails to sense the table height.

### Detailed context

We looked into the parameters of the ARE demo but we could not make it work properly (as it was working for other iCubs)
We can`t help thinking that there might be an issue with the force torque sensor we are using for the demo.

reafrancesco, 2023-08-01:
Grazie @pattacini
I will try to approach this strategy according to your suggestion. I will report here the results of my tests since it might be of help for the rest of the community.
Many thanks again.

sgiraz, 2023-09-28:
Hi @reafrancesco,

I see that there hasn't been any activity on this issue for a while. 
Do you still need support or have something to report here? Otherwise, I would close this issue.

reafrancesco, 2023-09-29:
No it has not been solved. I wanted to ask you if you can check if the hardware is OK at least.

AntonioConsilvio, 2023-10-05:
With @reafrancesco we agreed that some tests on the state of the FT-sensor on the right shoulder might be necessary, which might compromise the demo.

Therefore, further information will be provided when possible.

AlessandroT99, 2023-11-07:
Hi, I understood talking with @AntonioConsilvio that the FT-sensor will be substituted in the following days, and the old one would be tested. If any issue will be found in it could be possible to know the wrong parameters in order to correctly reconstruct some already taken data? 
Thank you in advance!

AntonioConsilvio, 2023-11-07:
Hi @AlessandroT99, it is correct, we will replace the FT with a new one so you can continue with your testing; and we will test the old FT.
However, the intervention will be scheduled when possible, and you will receive updates on this.

cc @sgiraz

sgiraz, 2023-11-07:
Hi @AlessandroT99,

We would like to schedule an intervention for your robot during the week of November 20th. Please let us know if you have a preferred day during that week, assuming it is compatible with your tasks.

cc @AntonioConsilvio @Fabrizio69

AlessandroT99, 2023-11-07:
Hi @sgiraz,

For us could be better monday the 20th of November, is it possible according to your availabilities?

sgiraz, 2023-11-08:
> Hi @sgiraz,
> 
> For us could be better monday the 20th of November, is it possible according to your availabilities?

We confirm our availability for the 20th of November. If someone from your team could assist us during the intervention, it would be greatly appreciated 🙏🏻.

cc @AntonioConsilvio

sgiraz, 2023-11-08:
/reminder 17 of November Maintenance activity scheduled for next Monday (cc @Fabrizio69 @AntonioConsilvio)

AntonioConsilvio, 2023-11-22:
Hi @AlessandroT99 @reafrancesco, Me and @AntonioAzocar replaced the robot FT on the right shoulder (SN: 164) with a new one (SN: 082).

From the first tests it seemed to work properly, but I am waiting for your feedback.

cc @sgiraz

sgiraz, 2023-11-29:
I close this issue since the intervention is concluded.
Feel free to reopen if necessary.

AlessandroT99, 2023-12-12:
Hi everyone!

Finally we aimed to get some tests on the arm in order to evaluate the new sensor.
Up to know we have not yet tested the ARE application, but I begin to anticipate you a strange behavior.

I report in the following plot, the behavior of the working left arm F/T sensor, dumped during an experiment from the port `/icub/left_arm/analog:o`.

[image]

Now the same experiment with the same condition has been repeated with the right arm before and after the F/T sensor substitution, and the results are attached in the following.
* Before the substitution:
[image]

* After the substitution:
[image]

This data are not manipulated but are just read and plotted. We can see an improvement  but it seems that an important issue is still present. Have you any idea of what can cause this shifting in all the axis? 
I also have to underline that the signals are read during the experiment which is made just after iCub power on and with the sensor calibrated throught the command `calib all` in the `/wholebodydynamics/rpc:i` port.

Thank you!

AlessandroT99, 2024-01-02:
Hi everyone!

Since last update, during a different talk with @pattacini I had the chance to introduce to him the problem with the sensor.
He said that the above difference from the raw signal of the sensor of the arms could not be a problem, because it should be removed from the wholeBodyDynamics with the gravity compensation (if I am not wrong), so he suggest to try also to look at the output in time of the `/wholeBodyDynamics/involved_arm/cartesianEndEffectorWrench:o` port, and observe if, after the _calib all_ command, some arm movements and some time, getting back in the same pose of the calibration gives as result from the mentioned port a signal very close to zero. 
I made this trial In both arms and the error at the end of the process is around 0.2 N for each arm.

Does this test confirm the correct functioning of the new sensor or more other check or calibration are needed both from you and/or us? 
Also further news from the ARE application test will be provided soon.

By the way, I get also the chance to wish to you happy new year!

pattacini, 2024-01-02:
Hi @AlessandroT99 and happy New Year!

Your tests confirmed somehow that the new sensor is fully operational. Hence, you can go on.
The only problem is the bias[^1], which by the way is removed by WBD in your context, so no problem.

[^1]: A well-calibrated FT sensor should return the arm weight in stationary conditions, projected on its three axes, which are in turn oriented wrt the root frame according to the kinematic. However, as said above, this bias is removed by WBD.

AlessandroT99, 2024-01-02:
Perfect thank you @pattacini! 

So before considering the issue finally close, I will let you know the results of the test on ARE application with @reafrancesco as soon we made it!

AlessandroT99, 2024-01-15:
Hi all!
Here as promise the consideration about the ARE platform.
We made some tests and now the right arm is again sensible to find the table height.

Thank you for the support and advice!

closed 2024-01-15 by AlessandroT99