robotology/mesh-support #1602: `ergoCubSN000` right knee encoder not moving

mebbaid, 2023-07-18:
### Device name 🤖

`ergoCubSN000`

### Request/Failure description

While the robot was on, the `right_knee` joint moved by itself and then went into hardware fault showing the error message i2t limit exceeded. When @S-Dafarra tried to move it , the motor position reads/changes while the encoder value doesn't move.

### Detailed context

Here is the log file

[log_ergocub-torso_yarprobotinterface_right_knee.txt](https://github.com/robotology/icub-tech-support/files/12083392/log_ergocub-torso_yarprobotinterface_right_knee.txt)

S-Dafarra, 2023-07-18:
cc @CarlottaSartore @lrapetti

AntonioConsilvio, 2023-07-21:
Hi @S-Dafarra @mebbaid, have you had problems with these joints again?

cc @sgiraz @CarlottaSartore @lrapetti

mebbaid, 2023-07-21:
We will test this afternoon and confirm to you @AntonioConsilvio

mebbaid, 2023-07-25:
We have the same situation on the `torso_yaw` today. The joint moved by itself, giving i2t error and encoder stopped moving even when moving the joint in idle. Below is a picture and the associated log

[image]

[log_ergocub-torso_yarprobotinterface_torso_yaw.txt](https://github.com/robotology/icub-tech-support/files/12160165/log_ergocub-torso_yarprobotinterface_torso_yaw.txt)

AntonioConsilvio, 2023-07-25:
> We have the same situation on the `torso_yaw` today. The joint moved by itself, giving i2t error and encoder stopped moving even when moving the joint in idle.

Hi @mebbaid, we will look at it, but for the torso yaw problem It would be better to open a separate issue. 

By the way, do you think that the problem on the right knee is solved? If yes, we will proceed closing this issue.

cc @sgiraz @Fabrizio69 @S-Dafarra

mebbaid, 2023-07-25:
No worries.
I commented on the same issue since it is basically the same problem of the encoders also affecting the same type of joint (yaw angle in both cases). But we can open a separate issue, if it is still necessary

sgiraz, 2023-07-28:
Hi @mebbaid,

The problem in the right knee has been solved, so I am closing this issue.

> We have the same situation on the `torso_yaw` today.

I recommend you open a new issue to track the problem of the `torso-yaw` even if it is related to the same type of joint (yaw). From the point of view of maintenance is very important to have an issue for each intervention.

closed 2023-07-28 by sgiraz