robotology/mesh-support #1603: `ergoCubSN000` left ankle roll weird jumps

mebbaid, 2023-07-18:
### Device name 🤖

`ergoCubSN000`

### Request/Failure description

@S-Dafarra noted a weird sound coming from the left ankle. Using [yarp-openmct](https://github.com/ami-iit/yarp-openmct) to plot the ankle roll we noted the  following jumps of +-0.3 

[image]

### Detailed context

The left ankle roll moves

S-Dafarra, 2023-07-18:
cc @CarlottaSartore @lrapetti

AntonioConsilvio, 2023-08-01:
Hi @S-Dafarra @mebbaid, today I checked the electronical wiring of the magnetic sensor and it seemed in good condition and without problems.

Also, I have tested the joint and it doesn't seem to have any particular problems at a macroscopic level.

So when you will do your tests again, I would like to ask you if you could provide feedback on whether the problem persists and therefore it needs further investigation.

cc @maggia80 @pattacini @sgiraz

S-Dafarra, 2023-08-02:
The issue is still occurring. The jumps produce some noise that are audible in the following videos. They are also perceivable when touching the foot

https://github.com/robotology/icub-tech-support/assets/18591940/82c1b371-2c6d-45c0-ba1b-a7f63cc90488

https://github.com/robotology/icub-tech-support/assets/18591940/d39c37a9-df84-4411-b588-1b134227d272

fbiggi, 2023-08-03:
General check of the screws
the mechanical strikes were stuck (they were loose)
check performed together with @AntonioConsilvio 
The condition seems to have improved
We look forward to your feedback after your tests

[image]

https://github.com/robotology/icub-tech-support/assets/20420331/c442ea9d-602c-46ba-b20d-8b3aa9798dae

@maggia80 @Fabrizio69

AntonioConsilvio, 2023-08-24:
Hi @S-Dafarra, We replaced the motor and put ferrite on the motor phases (LL_M5) and AMO cables (LL_S4 and LL_S5). for maintenance service due this issue:

- https://github.com/robotology/icub-tech-support/issues/1619

During my tests, I no longer detected the strange noise on the foot, so please give us feedback after your tests if the motor jumps or not.

cc @sgiraz @mebbaid @lrapetti @GiulioRomualdi

sgiraz, 2023-09-07:
Hi guys,

Do you have any feedback to share here?

cc @S-Dafarra @mebbaid @GiulioRomualdi

mebbaid, 2023-09-07:
> Hi guys,
> 
> Do you have any feedback to share here?
> 
> cc @S-Dafarra @mebbaid @GiulioRomualdi

We did not notice the jumps on the `left_ankle_roll`, however, to be perfectly honest we didn't pay particular attention during the last tests to catch it. In case any update I will drop it here. Meanwhile I would go with the same fix for the right
ankle