robotology/mesh-support #1604: [ergoCubSN000] Boards support broken

lrapetti, 2023-07-19:
### Device name 🤖

ergoCubSN000

### Request/Failure description

The support of the boards on the torso back is broken,  plus one of the encoder cable is broken (possibly other cables got broken).

[image]
[image]
[image]

### Detailed context

The robot felt on the back hitting the crane.

AntonioConsilvio, 2023-07-19:
hi @lrapetti @CarlottaSartore, me and @GiacomoVassallo13 replaced all damaged holder boards.

Then, I fixed the damaged cable (WS0).

The robot has been tested and now works properly.

cc @sgiraz

sgiraz, 2023-07-20:
Excellent job, everyone! 🚀

closed 2023-07-20 by sgiraz