robotology/mesh-support #1605: iRonCub3 (iCubGenova09): cable broken on the left shoulder

gabrielenava, 2023-07-19:
### Device name 🤖

iCubGenova09

### Request/Failure description

One of the cables on the left shoulder got detached and needs to be crimped again.

### Detailed context

cc @vpunithreddy 

### How does it affect you?

we cannot perform iRonCub experiments

AntonioConsilvio, 2023-07-19:
Hi @gabrielenava, the cable LA_S1 was damaged. I fixed it and tested.

The robot shoulder works properly now, however give us a feedback after your experiments.

cc @sgiraz

AntonioConsilvio, 2023-07-24:
Hi @gabrielenava, I guess that the issue has been solved. So, I procede closing the issue. 

Feel free to reopen if necessary.

cc @sgiraz

closed 2023-07-24 by AntonioConsilvio