robotology/mesh-support #1606: [ergoCubSN000] left wrist encoder not reading

lrapetti, 2023-07-19:
### Device name 🤖

ergoCubSN000

### Request/Failure description

The left wrist joint encoder is always showing `0`.

### Detailed context

The robot start, but the left wrist does not moved. When checking on the yarpmotorgui we can observe that the joint reading is always zero while the motor reading is changing.

lrapetti, 2023-07-19:
After moving it (by @AntonioConsilvio) and restarting the motors, the problem is no longer occurring. I think we can close this.

sgiraz, 2023-07-20:
> After moving it (by @AntonioConsilvio) and restarting the motors, the problem is no longer occurring. I think we can close this.

Ok, so I proceed to close this issue.

cc @AntonioConsilvio

closed 2023-07-20 by sgiraz