robotology/mesh-support #1611: [ergoCubSN000] Wrong date in ergocub-torso

GiulioRomualdi, 2023-07-24:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

@isorrentino noticed that the date of the ergocub-torso is pretty old. This turns in several cmake warnings and unnecessary recompilation of the code in the robot-torso

### Detailed context

This is the output of the date in the ergocub setup. Only the ergocub-torso has a wrong date 

[image]

### Additional context

cc @simoneMic @S-Dafarra

SimoneMic, 2023-07-26:
This issue is also a problem for the ROS2 topic detection and the TF tree, I'll add just a quick note:
As mentioned [here](https://github.com/icub-tech-iit/ergocub-software/issues/142#issuecomment-1620444246) the solution would be to synchronize all the PCs to the `10.0.2.1` server. I have done it on `ergocub-laptop`, but on the torso and head I don't have the root privileges.
In the end it takes only 2 minutes following [this guide](https://www.linuxbabe.com/linux-server/how-to-enable-etcrc-local-with-systemd), just take the `rc-local` from the laptop and do the same thing on both torso and head.

If another approach is used, please do the same thing on all the machines for consistency. Thanks.

SimoneMic, 2023-07-26:
By the way the quick fix is to run `sudo ntpdate -b 10.0.2.1` once (you can reverse search it in the terminal since I use it quite a lot)

maggia80, 2023-07-26:
@davidelasagna could it be related to the BIOS battery?

davidelasagna, 2023-07-26:
@maggia80 it could but if I'm not mistaken We changed the battery just before ICRA,  I've checked the torso and after the reboot the date and time changed form the correct one to one in the past. I've set up a workaround creating a startup service that synchronize the time with iiticubws079.
In any case a check to the CMOS battery should be done sooner or later.

sgiraz, 2023-07-28:
Hi @GiulioRomualdi, could check if the torso XSENSE IMU works expected? If not, maybe we have to check the CMOS battery, otherwise the problem may be solved after the intervention of @davidelasagna.

I put this issue in review waiting for your feedback.

sgiraz, 2023-08-01:
Hi @GiulioRomualdi,

> I tried the imu but it doesn't start

Ok, we will check possible issues around COMM-Express and its carrier board.
To do this, we'll have to remove the torso cover and unmount the COMM-Express for further inspection.

🎯 Possible action points:
1. check wirings, connectors, connections, and CMOS Battery
2. check the BIOS version
3. check the BIOS configuration
4. try to update the BIOS 
5. replace the COMM-Express board with another one

Stay tuned for further updates!

cc @maggia80 @Fabrizio69 @Gandoo @AntonioConsilvio

GiulioRomualdi, 2023-08-01:
Hi @sgiraz thank you for the update 😃

Nicogene, 2023-08-08:
Alongside @davidelasagna this morning we investigate this [problem](https://github.com/robotology/icub-tech-support/issues/1611#issuecomment-1658372332) but we didn't manage to make the imu work either w/ or w/o BIOS settings.
We tried also to use the cable and the imu @martinaxgloria used on the bench but we still get the same error.
We will continue investigating the setup on the bench.

S-Dafarra, 2023-08-08:
> Alongside @davidelasagna we managed to solve both IMU and date problems.

Also the date issue was caused by the version of ``yarp-device-xsensmt``?

Nicogene, 2023-08-08:
> > Alongside @davidelasagna we managed to solve both IMU and date problems.
> 
> Also the date issue was caused by the version of `yarp-device-xsensmt`?

It is not clear what was the cause of that issue, we saw that when the cable was attached there were strange glitches in the bios, like someone was pressing keys, we supposed that was due to faulty communication on serial.
We noticed that the cable was a little bit damaged, after fixing it these glitches disappeared, but we are not 100% that was the cause

SimoneMic, 2023-08-24:
> Alongside @davidelasagna we managed to solve both IMU and date problems. Testing on bench we verified that everything was working flawlessly, then the only difference were the software. On the com-express on bench we have the latest master of `yarp-device-xsesnmt` ([robotology/yarp-device-xsensmt@7a2de59](https://github.com/robotology/yarp-device-xsensmt/commit/7a2de59ad2343be23fce86db850215a4f18d9e3e)), while on the com-express of ergocub it was on [`v2.2.0`](https://github.com/robotology/yarp-device-xsensmt/releases/tag/v0.2.2) tag that is of November 2022. Updating to the latest master made **the waist imu work also without changing the bios setting that right now is using the default ones.**

Just for clarification, is the configuration on the robot already aligned and working? Can we use the torso imu?
Thanks

sgiraz, 2023-08-24:
Hi guys,

> It is not clear what was the cause of that issue, we saw that when the cable was attached there were strange glitches in the bios, like someone was pressing keys, we supposed that was due to faulty communication on serial.

After a deep inspection made by @davidelasagna and @AntonioConsilvio We can confirm that the problem has been solved by configuring the BIOS COM port settings properly. 

> On the com-express on bench we have the latest master of `yarp-device-xsesnmt` [...]

As per this, I suggest to check often the alignment among the repos inside the `robotology-superbuild` in order to avoid unwanted misalignments and issues.

> Just for clarification, is the configuration on the robot already aligned and working? Can we use the torso imu?
> Thanks

Afaik the robot is already aligned and you can use the torso-imu.

Considering the problems mentioned in this issue solved, I proceed to close it.

closed 2023-08-24 by sgiraz