robotology/mesh-support #1613: ergoCub 1.0 S/N:000 : Inaccurate Joint Tracking

Woolfrey, 2023-07-24:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The low level joint control on the arms needs tuning to improve trajectory tracking.

### Detailed context

The joint tracking on the arms has up to 4 degrees of error on most joints. This is causing issues with the Cartesian control since the pose of the hands cannot be guaranteed. The right wrist has large errors - up to 20 degrees - though this appears to be a recent problem that we didn't observe a month ago.

### Additional context

We recorded the tracking error for the joints tracking a joint-level trajectory:

[image]

[image]

We then recorded the joint tracking when running a Cartesian controller. The errors are much larger. Cartesian feedback is used to try and make sure the hands stay together when grasping with 2 hands, but since the low level joint tracking is innacurate this is causing the robot to overcompensate:

[image]

[image]

Here is a video of the 2-handed grasping **before** ICRA 2023 and **without** Cartesian feedback on the position error of the hands. The ergoCub drops the object because the hands do not move together precisely because of the low-level joint tracking problem:

https://github.com/robotology/icub-tech-support/assets/62581255/62ec6fc4-258c-4754-a243-615d86ec4164

<br>

Here is a video **after** where feedback on the hand positions was used to try and keep them together. The robot overcompensates because the joints are not tracking accurately and the error on the hand poses increases instead of decreasing:

https://github.com/robotology/icub-tech-support/assets/62581255/3d66a1a1-aed1-476b-9b8c-b29899f4d465

### How does it affect you?

Myself @woolfrey and Vignesh @vigisushrutha23 need accurate Cartesian control to demonstrate research in bimanual manipulation as part of the ergoCub project.

High-level feedback control to correct the pose error of the hands is also not possible unless the underlying joint tracking can be guaranteed (as shown in the video).

@xEnVrE : FYI

maggia80, 2023-07-25:
@Woolfrey looking at the plots there are some issues in the wrist roll, right arm. We will investigate and fix it.

For what concern joint tracking accuracy, we used to provide the robot with "default" values of PID parameters. The tuning of the PID since it's application dependent is left to the user. What works for you may not work for some other users and vice-versa.

AntonioConsilvio, 2023-07-26:
Hi @Woolfrey, The right wrist roll is not able to reach negative values. Today me and @fbiggi have tightened the screws to reduce the backlash of the wrist, tomorrow we will check the motor belts (although we believe they are already properly tightened).

If the problem on the roll is a mechanical malfunction, then these are the two steps that can help. 

cc @maggia80 @sgiraz @Fabrizio69 @fiorisi

AntonioConsilvio, 2023-07-28:
Today I replaced the damaged board AMC-BLDC 30B2 (code 14962.C S/N: 013), with a new one (S/N 018). 

But that was not enough, because I found the roll motor burned out (this is the motor connected to the damaged AMC-BLDC board).

We will replace the motor in the first days of next week.

cc @maggia80 @sgiraz @Fabrizio69

AntonioConsilvio, 2023-08-01:
With the help of @Gandoo and @fbiggi, we replaced the burnt-out motor and now the right wrist works well.

https://github.com/robotology/icub-tech-support/assets/114915464/a000524f-1e8a-441e-94f1-ffc4bc3252d8

The PR updating the configuration files is available here:

- https://github.com/robotology/robots-configuration/pull/550

cc @maggia80 @sgiraz @Fabrizio69 @S-Dafarra

vigisushrutha23, 2023-08-01:
> Hi @Woolfrey @vigisushrutha23, after the motor replacement you should have better tracking on the right wrist joints, I don't know if it will completely solve the tracking problems, but you should see an improvement.

Thanks @AntonioConsilvio . We will run tests and post the results tomorrow so that if there are no further problems, we can close the issue.

sgiraz, 2023-08-03:
Hi @vigisushrutha23, 

I guess the tests went well, right? 
If yes, we proceed to close this issue.

vigisushrutha23, 2023-08-03:
> Hi @vigisushrutha23,
> 
> I guess the tests went well, right? If yes, we proceed to close this issue.

We will be doing them starting from 2pm. Sorry for the delay. We will post the plots ASAP so that the issue can be closed if there are no problems. Thanks.

Woolfrey, 2023-08-17:
@AntonioConsilvio @vigisushrutha23, much better results now.

The left wrist still has better tracking than the right wrist, but compared to before the magnitude of the tracking error is much smaller.

We are still noticing strange behaviour when grasping with 2 hands in Cartesian mode. Part of this problem is likely due to my controller trying to over-correct the hand positions.

**Joint Control Mode**

[image]

[image]

**Cartesian Control Mode**

[image]

[image]

sgiraz, 2023-08-29:
> The left wrist still has better tracking than the right wrist, but compared to before the magnitude of the tracking error is much smaller.
>
> We are still noticing strange behaviour when grasping with [...]

Hi @Woolfrey,

In light of the above, can we consider the problem solved from our end?
If yes, I proceed to close this issue.

cc @AntonioConsilvio

Woolfrey, 2023-09-01:
Hi @sgiraz, @vigisushrutha23 and I ran some more testing today with the 2-hand grasping.

I removed the grasping constraints in my controller that was causing it to over-correct on the wrist tracking error and the motion is similar to what we had before ICRA:

https://github.com/robotology/icub-tech-support/assets/62581255/2ac924c9-804e-4220-a1df-8f1e9baf1014

<br>

The wrists still aren't aligned correctly so this makes it difficult to grasp an object with 2 hands. The tracking error is _slightly_ higher than the last time, but still roughly the same magnitude:

[image]

[image]

The left wrist has a lot of compliance, which is something we have noticed for several months now:

https://github.com/robotology/icub-tech-support/assets/62581255/ab9f5b96-f909-4238-bcf5-0754161198ea

Here is another test we did with code I had written to optimize the arm configuration in Gazebo:

[image]

The robot slowly stretches out the right arm.

When running this code on the real robot the wrist rotates around indefinitely (i.e. it is unstable).

https://github.com/robotology/icub-tech-support/assets/62581255/8110115c-e204-4f68-8a04-4f329aa83374

<br>

It is possible that my control code needs some refining to work with the real robot, but given the wrist issues this could also be an explanation.

vigisushrutha23, 2023-09-10:
Today @mebbaid and i ran some experiments which used a walking controller receiving references from the @Woolfrey bimanual module. We found four joints having significant tracking errors namely l_shoulder_pitch, l_shoulder_roll, r_shoulder_pitch, r_shoulder_roll. The plots of desired vs measured are below. 
L_SHOULDER_PITCH
[image]

L_SHOULDER_ROLL
[image]

R_SHOULDER_PITCH
[image]

R_SHOULDER_ROLL
[image]

The somewhat similar nature of the errors this time points to the likely problem being low-level PID tuning which we will attempt to do for these four joints.

xEnVrE, 2023-09-10:
Cab I ask what is the unit of measurement in the plots? Degs or rad?

vigisushrutha23, 2023-09-10:
> Can I ask what is the unit of measurement in the plots? Degs or rad?

Radians

vigisushrutha23, 2023-09-18:
As of repeated testing last week, the only joints that give rise to errors are the 6 of the shoulders but I believe that will be taken care by re calibration of the PID by @mebbaid later this week or the next . So this issue can be closed. I will be opening a new issue for an issue with the left wrist. Thank you all once again for all the fixes.

sgiraz, 2023-09-25:
Hi @vigisushrutha23,

Thanks for your feedback. I proceed to close this issue then.

cc @AntonioConsilvio

closed 2023-09-25 by sgiraz