robotology/mesh-support #1615: Left wrist loose

S-Dafarra, 2023-07-25:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The left wrist appears to be very loose, with mechanical backlash.

### Detailed context

One possible cause (or a consequence) seems that the cable going to the hand can get stuck in the mechanism while the wrist is moving.

In the video below, the wrist is position controlled and then the cables get stuck inside the mechanism. After stopping the video, we were able to unlock the cables after performing some movements with the wrist.

https://github.com/robotology/icub-tech-support/assets/18591940/21d90da5-55a6-4409-8519-e5c8219ce311

AntonioConsilvio, 2023-07-26:
Hi @S-Dafarra, me and @fbiggi have tightened the screws to reduce the backlash of the wrist, tomorrow we will check the motor belts (although we believe they are already properly tightened).

These two steps are the only two interventions we can perform on the mechanism to reduce its backlash.

About the electronics cables routing, the cables should not induce backlash in the mechanism, however if we find a way to perform the cable routing in a way that is less invasive we will apply it.

cc @sgiraz @maggia80 @Fabrizio69 @fiorisi

AntonioConsilvio, 2023-07-28:
Hi @S-Dafarra, today me and @fbiggi overhauled the entire wrist mechanism. We tightened the wrist screws with the use of Loctite, checked the belts and also tightened the screws and glued the motor pulley that was producing more backlash.

All these interventions are the best that can be done to reduce wrist backlash on this type of mechanism.

The wrist has been reassembled and tested. It is functioning correctly. 

cc @maggia80 @sgiraz @Fabrizio69 @fiorisi

closed 2023-08-02 by AntonioConsilvio