robotology/mesh-support #1616: iRonCub3 - WiFi antenna adapter of icub-head no.2 (dedicated for cameras) is broken

HosameldinMohamed, 2023-07-25:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The WiFi antenna adapter used to communicate with the 2nd icub-head (dedicated for the cameras) was broken during running some motion sequences.

### Detailed context

Here's a picture:
[image]

CC @gabrielenava 

### Additional context

Please refer to the system architecture in https://github.com/icub-tech-iit/tickets/issues/2812#issuecomment-1580614256

I think it's the antenna of the `Xavier NX` board.

maggia80, 2023-07-31:
@gabrielenava we have one spare. It's on my desk. 

We need to buy some more spare parts: 

https://www.amazon.it/TP-Link-Archer-T3U-Adattatore-Dual-Band/dp/B07M69276N/ref=asc_df_B07M69276N/?tag=googshopit-21&linkCode=df0&hvadid=279793429868&hvpos=&hvnetw=g&hvrand=7981661490014642711&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1008337&hvtargid=pla-661383826758&th=1

I'll open a ticket.

maggia80, 2023-07-31:
@HosameldinMohamed if you come to my desk you can take it.

HosameldinMohamed, 2023-07-31:
Hi @maggia80, we connected the antenna but we are still not able to ping the head,, is there some configuration needed?

We see a blue light slowly blinking on the antenna adapter.

HosameldinMohamed, 2023-07-31:
The strange thing is that we made sure that all the cables are well attached,, we got a bliking blue light,, but no connection,, then we restarted the head and we never got the blinking light again.

I connected the antenna to my laptop and ran `sudo lshw -C network` but I see only my physical network adapters, and not the USB one,,, and no blinking light either..

maggia80, 2023-07-31:
Could the cable connecting the adapter and the carrier board be damaged as well?
I got it from @davidelasagna and it should work out of the box.

davidelasagna, 2023-07-31:
@HosameldinMohamed in a laptop that already has a wifi adapter it may be not visible. On the robot probably it has took another  interface name eg: "wlan1" instead of "wlan0" and since the configuration is for wlan0 it has the default config.

DanielePucci, 2023-08-01:
Probably @davidelasagna you may give some support for debugging this? Thanks

davidelasagna, 2023-08-02:
Yesterday along with @HosameldinMohamed  and @davidegorbani  I investigated  the issue. 
 As predicted the wlan interface had a different name. ssh-ing first on `icub-head` (10.0.0.2) I've been able to jump on the `icub-head-cam` via ethernet (10.0.2.3) and change the interface name using `nmtui`.
But We've encountered  another issue. As soon as We plugged the USB antenna the `icub-head-cam` crashed or went in a frozen status and it became stuck and unreachable.
Running `dmesg` on the NX board there was a recurrring error:  `soctherm: OC ALARM 0x00000002`
this seem to be related to an underpower issue. 
We've notice that by unplugging the USB-C cable that goes from the NX to the USB antenna the issue didn't occurred,
Moreover We tried using a USB-A to USB-C adapter and connecting the antenna and everything was working fine.
I think that this issue could be caused by  USB-C cable that could be damaged somehow. I've delivered it this morning to @sgiraz  for further checks.

sgiraz, 2023-08-03:
Hi guys,

@Gandoo started the maintenance activity yesterday morning. If everything will be ok, the robot will be ready by Monday 7th Aug.

cc @davidelasagna

sgiraz, 2023-08-21:
Hi @HosameldinMohamed,

I know that @Gandoo has changed the flat cable. Did you replace the antenna, right?

HosameldinMohamed, 2023-08-21:
Hi @sgiraz

We just tried to ping `icub-head-cam` via the wifi antenna, but we couldn't do it. The board seems to be not ping-able.

Probably this issue is not even related to the antenna but to the board as a whole. Because we also tried what @davidelasagna has done in https://github.com/robotology/icub-tech-support/issues/1616#issuecomment-1661717047

> ssh-ing first on `icub-head` (10.0.0.2) I've been able to jump on the `icub-head-cam` via ethernet (10.0.2.3) 

But we couldn't access it. Probably more investigation is needed for the board.

Actually, last Friday I was able to access `icub-head-cam` with the wifi antenna.

sgiraz, 2023-08-22:
Hi @HosameldinMohamed,

As we saw together yesterday, the issue was caused by the repeated restarting of the NVIDIA Jetson Nano board mounted on the head.
After reconnecting the Nano's board power cables, the robot's head became po gable again.

@davidelasagna Thank for the support.

Closing.

HosameldinMohamed, 2023-08-22:
Thanks all!