robotology/mesh-support #1618: RobotInterface doesn't start because of the Right arm board is not able to discover CAN board

SimoneMic, 2023-08-04:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The right arm board `right_arm-eb30-j4_6 IP 10.0.1.30` is not able to discover one of its CAN boards, so the robot is not able to start.
I have checked the firmware updater and all the boards were in application.
After several switching the motors on and off, the issue persisted.
I am attaching an extract of the console log where the errors happened:
[errorlog_04-08-2023.txt](https://github.com/robotology/icub-tech-support/files/12261732/errorlog_04-08-2023.txt)

### Detailed context

While navigating the robot with the crane, it suddenly entered in hardware fault (from the right arm) and collapsed on itself due to the power cut. The robot didn't hit anything, nor event crashed on itself without any relevant collisions.
After switching it back on, the robotinterface wasn't starting. The file that I used is `ergocub_wbs_ros2.xml`

Further notes:
The robot was in battery mode and charged with both the power supplies on.
No particular errors for the hardware fault.

AntonioConsilvio, 2023-08-07:
Ciao @SimoneMic Ho trovato il motore M0 del polso destro e la scheda 30B0 bruciati.

Pianificheremo la sostituzione

cc @sgiraz

AntonioConsilvio, 2023-08-07:
@SimoneMic regarding random shutdowns of the robot's boards (and therefore the motors), there is already an [issue](https://github.com/robotology/icub-tech-support/issues/1614) we are working on.

cc @sgiraz

AntonioConsilvio, 2023-08-16:
Hi @SimoneMic, we replaced the burnt-out motor and the burnt-out board. 

The robot has been tested and now the right wrist is working properly.

https://github.com/robotology/icub-tech-support/assets/114915464/2d780326-3f07-4231-bbac-3f2ed2d6be77

This is the PR that updates the configuration file (the robot is already updated):

- https://github.com/robotology/robots-configuration/pull/554

cc @sgiraz @mfussi66

closed 2023-08-16 by AntonioConsilvio