robotology/mesh-support #1619: Left ankle roll moved fastly to the joint limit and started to smell burnt going into hardware fault

isorrentino, 2023-08-18:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

While using the right leg of the robot the left ankle roll moved to the limit without any input and started to smell. Unfortunately, I did not have the log file (I forgot to run the `yarplogger`) and now I'm afraid to turn the robot on again, so I cannot take the log.

### Detailed context

As per the description.

AntonioConsilvio, 2023-08-23:
Hi @isorrentino, after some tests we realised that the motor of the left ankle roller is burnt out. 

Tomorrow morning we will replace it.

cc @maggia80 @fbiggi @sgiraz

AntonioConsilvio, 2023-08-24:
Hi @isorrentino, with the help of @fbiggi and @Gandoo, we replaced the burnt out motor with a new one.

The robot has been tested and it works properly now.

https://github.com/robotology/icub-tech-support/assets/114915464/d33ad346-0960-4b58-af01-21da67914286

⚠️ Remember to update robots-configuration with the new calibration value

Check the PR:

- https://github.com/robotology/robots-configuration/pull/557

cc @sgiraz @SimoneMic

closed 2023-08-24 by AntonioConsilvio

isorrentino, 2023-08-24:
Thank you all!