robotology/mesh-support #1620: iCubGenova09 (iRonCub3) S/N:000 - Right elbow motor cable is broken

HosameldinMohamed, 2023-08-21:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The cable that connects the right elbow motor is broken, see the image below.

### Detailed context

I removed the cover of the right arm to access the cable of the FT sensor. When I put the cover back the robot didn't start. When I removed it again I saw this broken cable.

### Additional context

[image]

### How does it affect you?

It blocks iRonCub experiments.

CC @gabrielenava

HosameldinMohamed, 2023-08-21:
I updated the description, but I can't update the points..

sgiraz, 2023-08-21:
Hi @HosameldinMohamed,

@AntonioConsilvio and @AntonioAzocar will try to fix it as soon as possible (today).

AntonioConsilvio, 2023-08-22:
Hi @HosameldinMohamed, I fixed the broken cable (RE3).

Here the video of the test:

https://github.com/robotology/icub-tech-support/assets/114915464/3df16141-1ac7-44b6-80df-fd05f0b1be45

cc @sgiraz

closed 2023-08-22 by AntonioConsilvio