robotology/mesh-support #1621: iCubGenova09 (iRonCub3) S/N:000 – Right arm yaw joints suddenly moved with no change in joint angle

HosameldinMohamed, 2023-08-22:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The yaw joint of the right arm moved suddenly without sending a command to move it. What could have triggered this issue is loading the right arm with a heavyweight (9kgs).

It happened only twice and in the second time no weights were attached to the robot.

### Detailed context

When performing tests with the robot yesterday, while attaching heavy weights, the yaw joint of the right arm suddenly moved, but the joint angle did not detect that change. The offset persisted until restarting the robot, then it didn't happen again.

This morning a similar motion happened again with the same joint, but the robot was not loaded. Running the calibration of the joint removed the offset.

Apart from the two cases, the issue did not happen again, and we are not able to reproduce it at the moment.

HosameldinMohamed, 2023-08-22:
[image]

It happened again right after restarting the robot 
It moved to this location and it's considered the new 0deg position..

Recalibration ristored the correct position..

HosameldinMohamed, 2023-08-22:
It started to occur more frequently, 

This time the joint position jumped to a value outside the range, and due to the `i2t` limit we had a hardware fault.

[image]
[image]

DanielePucci, 2023-08-22:
I think we got it @AntonioConsilvio @sgiraz Probably this may also be the cause of https://github.com/robotology/icub-tech-support/issues/1619

https://github.com/robotology/icub-tech-support/assets/5045846/4e25783c-c070-478b-870b-cd94a60e0ec5

CC
@isorrentino @GiulioRomualdi

AntonioConsilvio, 2023-08-23:
Hi @HosameldinMohamed @gabrielenava @DanielePucci, me and @AntonioAzocar replaced the cable RA_S2 (that connect the AMO sensor to the EB4 ems board). 

The robot has been tested and it is working properly now. 

cc @sgiraz

closed 2023-08-23 by AntonioConsilvio