robotology/mesh-support #1622: iCubGenova09 (iRonCub3) S/N:000 – Right arm FT sensor readings freeze when applying high loads

HosameldinMohamed, 2023-08-22:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

Some days ago, we noticed that the right arm FT changed it's behavior, and the change was significant. Moreover, when applying a high load on the sensor, the readings freeze, which didn't happen a few months ago.

### Detailed context

Details of the FT changing its behavior in https://github.com/ami-iit/element_ft-nonlinear-modeling/issues/83

Details of freezing the FT readings in https://github.com/ami-iit/element_ft-nonlinear-modeling/issues/89#issuecomment-1688042147

AntonioConsilvio, 2023-08-23:
Hi @HosameldinMohamed, me and @fbiggi replaced the FT sensor with a new one.

We tested the old FT sensor (SN: 080) and found Channel 4 saturated. We tried to solve the problem without success. 

We then replaced the FT sensor with a new one (SN: 090). This FT has been tested and it works

However, please give us feedback after your tests.

cc @maggia80 @sgiraz @gabrielenava

AntonioConsilvio, 2023-08-23:
Thank you @HosameldinMohamed for the feedback!

Since the data plot is good, I will proceed with closing the issue.

If the problem recurs, feel free to reopen the issue.

Cc @maggia80 @sgiraz

closed 2023-08-23 by AntonioConsilvio