robotology/mesh-support #1625: Fault button does not work

gabrielenava, 2023-08-24:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

We tried to start iRonCub using the fault button we normally have for experiments. It did not start, and in the logger we saw the message "external fault button pressed".

After several failed attempt, we switched to another fault button and the robot finally started. We concluded that our fault button is faulty.

### Detailed context

cc @vpunithreddy @davidegorbani

AntonioConsilvio, 2023-08-25:
Hi @gabrielenava I check the cables inside the jack connector and they was fine:

[image]

We tested the cable with the tester and the fault button behaviour is correct:

The button makes sound when the circuit is closed (button not pressed), and does not make sound when the circuit is open (button pressed).

https://github.com/robotology/icub-tech-support/assets/114915464/34306ed7-593e-486b-9a8f-c2ca1927710d

So, According to the test, the fault button works, however, I would wait for the availability of the robot on the roof to test the fault button on the robot again.

cc @sgiraz

gabrielenava, 2023-09-01:
>So, According to the test, the fault button works, however, I would wait for the availability of the robot on the roof to test the fault button on the robot again.

we tested again the fault button on the robot this week: same result, the robot does not start with error message "external fault button pressed". We tried with two other different fault buttons and they both work fine. At this point, I removed the fault button which is giving problems and delivered it to @AntonioConsilvio for further investigations.

AntonioConsilvio, 2023-09-14:
Hi @gabrielenava, I tried your fault button on another robot and it works correctly. So, maybe the problem is the FED board which might be loose. 
In fact behind the back cover there is the FED board (green in the picture below), you should check that the white holder and the fault jack are in good condition and also that the board is screwed in tightly (purple in the picture below).

[image]

cc @sgiraz

gabrielenava, 2023-09-19:
thank you @AntonioConsilvio 

>, you should check that the white holder and the fault jack are in good condition and also that the board is screwed in tightly (purple in the picture below).

we will check, let us know if and when you might be available for joining us

sgiraz, 2023-10-09:
Hi @gabrielenava, I guess the problem should be solved (probably the solution is to try to push the jack of the fault button with a bit force more than usual in order to avoid false contact due to the missing screw at the bottom of the support).\

Closing as solved.
Feel free to re-open if necessary.

closed 2023-10-09 by sgiraz

gabrielenava, 2023-10-10:
>Hi @gabrielenava, I guess the problem should be solved (probably the solution is to try to push the jack of the fault button with a bit force more than usual in order to avoid false contact due to the missing screw at the bottom of the support).

yes, this was the solution in the end! the fault button turned out to be ok