robotology/mesh-support #1627: pinkie finger of left arm wiring broken and index finger touch skin signal line detach from the pcb

jimingda, 2023-08-31:
### Robot Name 🤖

iCubShanghai01 S/N:049

### Request/Failure description

Hi, the end wire of pinkie finger of left arm is broken, as showed in pic below.
I think its a big problem, after checked the below instructions, it seems a complicated one.
My another concern is that will this infect the motion of other fingers on left arm? I am not sure if other fingers are normal, but from the appearance they are ok.
 
https://icub-tech-iit.github.io/documentation/icub_tendons/lower_arm_V2/
[image]
[image]

### Detailed context

As for touch signal line of finger index of left arm, as showed, i think it's easier to handle, maybe welding it back to the pcb.
 
[image]

sgiraz, 2023-09-11:
Hi @jimingda,

> Hi, the end wire of pinkie finger of left arm is broken, as showed in pic below.
> I think its a big problem, after checked the below instructions, it seems a complicated one.

The tendons of the fingers are the most challenging part to repair or replace. If you agree, I would recommend that you send us the left lower arm of the robot so that we can fix all of the problems mentioned above (see the instruction above to remove the forearm from the robot).

> My another concern is that will this infect the motion of other fingers on left arm?

The pinky joint couples the little finger and the ring finger. All the other fingers should work independently.

---

### How to remove the forearm from iCub 2.7

1. Remove the upper arm covers.

2. Remove the 4 screws (circled in red) of the steel support with the electronics board (we need the electronics).

[image]

3. Remove the following 3 screws (circled in red):

[image]

4. Detach all the connectors that connect the left lower arm with the left upper arm

--- 

>**Important**
> Before shipping, make sure that the unmounted lower arm looks like the following:

[image]

cc @AntonioConsilvio @Fabrizio69 @GiacomoVassallo13

jimingda, 2023-09-12:
Thanks @sgiraz for your detailed instructions.

`1. Remove the upper arm covers.`
As my picture of upper arm cover, there are two kinds of screws, small ones and bigger ones. 
From my understanding, the first step is removing the small screws to unmount the skin, and second step is to remove the bigger screws and then unmount the cover?

[image]

Could you provide the recipient information if we succeed to unmount the left lower arm and send it to you.

jimingda, 2023-09-18:
Thanks @sgiraz 

Shall i need to unscrew these internal screws? And need i remove the cover of these touch sensors before unmount the plastic cover below the touch sensor.

I found these touch sonsors are connected to the board, is there any document about these connections, so i can take care when remove them.

[image]

sgiraz, 2023-09-19:
> Shall i need to unscrew these internal screws? And need i remove the cover of these touch sensors before unmount the plastic cover below the touch sensor.

Yes! Remove the 4 screws shown in the picture and **leave the touch sensors in place on the cover**. On the rear side of the cover, you'll find several `mtb4` boards. To disconnect the `mtb4` board from the `mc4plus eb24` board, simply detach the Molex CAN connectors located on the side of the `mc4plus eb24` boards.

>**Important**
>Take note of all the connections removed from the `mc4plus` boards in order to be able to re-mount everything correctly once the forearm is repaired. Every connection is labeled with a label and they can also be found at https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub2.7/pdf/iCub2.7_E3.6.3_14838_Logic.pdf (see pages 1 and 6)

[image]

> I found these touch sonsors are connected to the board, is there any document about these connections, so i can take care when remove them.

⚠️ Do not remove the touch sensors from the cover. It is not necessary and they are very fragile.

Let me know if you need further support!

jimingda, 2023-09-20:
After remove the 4 screws of the LA4 in the pic above, i can see three different connectors of sensor. And i also find LA0 as showed in  the pic below, so i also need to remove the screws in LA0?
[image]

jimingda, 2023-09-20:
After remove the single screw on LA0, i unmount the two covers of upper arm. 
From my understanding, what to do next is
1. Disconnect the black line connectors to the touch sensor board in the pic below.
[image]

[image]
3. remove the 4 screws on the mental board in the pic below.
[image]

4. Then the upper and lower arms can be separated

jimingda, 2023-09-21:
As the instruction in  https://github.com/robotology/icub-tech-support/issues/1627#issuecomment-1713425259

I succeed to 

1. Remove the upper arm covers.
2. Remove the 4 screws (circled in red) of the steel support with the electronics board (we need the electronics).

Still there are 3 connectors that connect the left lower arm with the left upper arm as showed in the pic below,
two of them are labeled E1 & E26, the last one labeled 12VA, which connected to EB24 EB25 EB26 seperatly and a big capacitor binded in the steel support. Shall i disconnect these three connections, including disconnect the capacitor?

[image]

jimingda, 2023-09-25:
Hi @sgiraz 

Follow the instructions i unmounted the left lower arm as showed below,  could you help to check it please?

Thanks in advance.

[image]
[image]

jimingda, 2023-10-16:
Hi @sgiraz 
Could you provide the email address of @Fabrizio69 
Thanks.

1. The weight and dimensions of the parcel: 1.6kg and 0.35m*0.25m*0.15m.
2. My address to pick up the goods:
Room 2214, Main Building, East Guanghua Building
No 30 Handan Road, Fudan University
200433 Yangpu District, Shanghai, China
Contact: Mingda Ji
Phone: +86 188 0196 7350
email: jimi1987@163.com

sgiraz, 2023-10-16:
Hi @jimingda,

Thank you for the parcel information. The person who will help you with the shipment is @andreaderito. Feel free to contact him at andrea.derito@iit.it

cc @Fabrizio69

andreaderito, 2023-10-17:
Hi @jimingda I confirm having taken this in charge - we continue this via e-mail exchange between me and you. Thank you for cooperating

sgiraz, 2023-12-07:
Hi @jimingda,

>My another question is how to ensure the the boards on arm like eb24 eb25 eb26 connected and works well as before, is there any method or tool to check it?

You can check them using the [FirmwareUpdater][1] application.

> It seems that the left hand didn't finish the calibration, because when i finished the iCubStartup demo, the left hand fingers didn't work well as right hand, and the final state of left hand as below:

We appreciate your effort in repairing the forearm. However, based on our experience, we recommend proceeding with the shipment of the forearm we initially planned.

>[!Important]
>We would ask you to re-pack the forearm maintaining the same weight and dimensions as before.

cc @andreaderito @AntonioConsilvio @Fabrizio69 @maggia80 

[1]: https://icub-tech-iit.github.io/documentation/icub_firmware/firmwareupdater/firmwareupdater/?h=firmware#discover

jimingda, 2023-12-11:
Hi @sgiraz 

Is it possible for us to calibrate the left hand on our own? We are uncertain about the typical duration for shipping and repairing the forearm. Can you provide more details based on previous experience?

Thanks in advance.

sgiraz, 2023-12-11:
> Is it possible for us to calibrate the left hand on our own? We are uncertain about the typical duration for shipping and repairing the forearm. Can you provide more details based on previous experience

Hi @jimingda,

We understand that you're having trouble calibrating your left hand and are unsure about the shipping and repair process. Troubleshooting complex electrical/mechanical issues remotely can be challenging, as there might be a variety of underlying causes that require physical inspection and manipulation ( (connectors not properly connected, broken wires, damaged boards, etc...).
To ensure that your left-hand returns to its optimal functionality as quickly as possible, we recommend that you proceed with the shipment for repair.

Based on our previous experience, the average time for repairing a forearm is typically between 1-2 weeks. However, the specific timeframe may vary depending on the severity of the issue and the availability of replacement parts.
@andreaderito may provide you with further details about the shipping time.

cc @Fabrizio69 @GiacomoVassallo13 @AntonioConsilvio

andreaderito, 2023-12-11:
Hi @jimingda concerning the shipping time, according to experience and also according to Shipping Company's indications:
0) Pick-up of the package (T0)
1) transit time 3-5 days
2) Customs' clearance (goods-in in temporary regime) Italian side: 2-4 days
3) Repair intervention - 5 - 10 days (i.e. 1-2 weeks, ref. @sgiraz ' s comment [above](https://github.com/robotology/icub-tech-support/issues/1627#issuecomment-1850367467))
4) Customs' clearance (goods out in temporary regime) Italian side: 2 days
5) transit time 3-5 days
6) Custom's clearance Chinese side: 3 - 4 days
7) T1 = Delivery at your premises: 1 day after clearance
So, in total, from T0 (day when the package is picked up from your lab) to  T1 the time span is estimated between min. 19 max. 29 working days.
We understand it is not short time, but all considered, and according to our decennial experience on iCub in all its versions from 15+ years ago till now, this is the best & most advisable way to proceed.
From our side, the contract with the shipping company for this shipping and for the management of the temporary import-export operation both China and Italy sides has been agreed, so now they will check the documents with both Customs Italy and China and I expect that during this week we may have the pick-up of the goods (i.e., T0).
cc @sgiraz @maggia80 @Fabrizio69 @GiacomoVassallo13 @AntonioConsilvio

andreaderito, 2023-12-12:
Hi @jimingda just to update you - we've passed all docs to the Shipping Company for their check with Customs both ITA and China sides so to smooth the shipping process and the temporary import-export operation - expected feedback in 24-48 hours, they said. Till soon for updates.
cc @sgiraz @maggia80 @Fabrizio69 @GiacomoVassallo13 @AntonioConsilvio

andreaderito, 2023-12-18:
Dear @jimingda just to update you, on Friday 15.12.2023 the Shipping Company (Italy side) gave OK to proceed to their colleagues in Shanghai, so I expect that they will contact you in short time to agree on finalisation of the shipping documents (packing list and proforma invoice) and to agree on date&time for the pick-up. 
cc @sgiraz @maggia80 @Fabrizio69 @GiacomoVassallo13 @AntonioConsilvio

jimingda, 2024-01-09:
Hi @andreaderito 

The company dealing with customs affairs is responsible for exporting and importing the iCub robot as well as the lower arm. They are requesting your confirmation of the repair agreement, which would be helpful.

Could you pls check the email, if it's ok, sign the printed document, and send back a scanned copy with your signature.

Thanks again.

andreaderito, 2024-01-09:
Dear @jimingda, yes, indeed there was an email dated yesterday in my junk box, now I recovered it - I confirm that I've passed the proposed Repair Contract to the 2 competent offices, 1 has already expressed OK but the second is still evaluating the document (it was Christmas & New Year holiday period so personnel was reduced). I'll update you asap. Thank you

andreaderito, 2024-01-15:
Hi @jimingda yes, we've just received the final OK to the document you proposed, I'm going to contact you via email for details

jimingda, 2024-02-16:
Hi  @andreaderito 

Have you received the pack? I guess that the lower arm has been transfered to IIT.

andreaderito, 2024-02-16:
Hi @jimingda yes, today I received information that parcel has arrived to Genoa, now we will proceed with the Customs documents and operations - we will update you as soon as the parcel is delivered to our premises - best regards

andreaderito, 2024-03-11:
Hi @jimingda we are processing the temporary import and IIT is about to pay the deposit of the requested money to the Italian Customs Office in order to grant for the temporariness of the import - my Central Offices say we should be able to receive the parcel within this week or max. beginning of next week - I'll update you - best regards

jimingda, 2024-03-14:
Hi @andreaderito 

I truly appreciate it, even though the process is a bit slow. Thank you, and I hope that the following steps can help speed things up.

andreaderito, 2024-03-19:
Hi @jimingda we should receive the parcel either on Friday 22.03 or on Monday 25.03.2024, this is the last piece of info we received by Central Offices. We will try to speed up the repair process. Thank you for your patience.

andreaderito, 2024-03-22:
Hi @jimingda the parcel has arrived at our premises, all seems OK (no damage on the outside) so next week our technical colleagues will open the parcel and inspect the goods / start repairing it. Thank you for your cooperation and kind patience - best regards
cc @sgiraz @AntonioConsilvio @maggia80 @Fabrizio69

jimingda, 2024-04-08:
Thanks @andreaderito 

I'm glad to hear that you've finally received the parcel. I was wondering about the progress of the repair process. Has it been completed?

AntonioConsilvio, 2024-04-11:
Hi @jimingda, we have just finished with the maintenance service, the forearm is working properly now!

## Calibration:

https://github.com/robotology/icub-tech-support/assets/114915464/a687e24c-14c6-4869-a466-f798fcd1ad03

## Skin:

https://github.com/robotology/icub-tech-support/assets/114915464/6ac71224-8e30-4045-8704-c8c03baf7380

We plan to start the shipping process next week!

cc @andreaderito

jimingda, 2024-04-17:
Thanks @AntonioConsilvio @andreaderito 

Please inform me once the shipping process commences. Additionally, could you confirm if Hellmann Logistics is still the designated shipping company? We need to prepare the necessary customs documents in advance.

andreaderito, 2024-04-18:
Hi @jimingda I'm going to send you an email to agree on details for the preparation of the shipping back - best
cc @AntonioConsilvio

jimingda, 2024-05-06:
Hi @andreaderito 
Is there any update on the shipping back issue?

andreaderito, 2024-05-06:
Hi @jimingda today (06.05.2024) I've received the offer for the shipping back, we're now working on the draft docs - I keep you informed - best
cc @AntonioConsilvio

andreaderito, 2024-06-26:
Thank you @jimingda for all your cooperation and support on this, and for your patience.

jimingda, 2024-07-05:
Hi,

I have some problem when restore the cables of left arm, where can i get instructions about this?

[image]