robotology/mesh-support #1630: Right hand finger lever detached

lrapetti, 2023-09-05:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

We notice the lever of the right finger pending.

[image]

### Detailed context

This probably happened when the robot abruptly shutter down due to https://github.com/robotology/icub-tech-support/issues/1614

fbiggi, 2023-09-06:
@sgiraz 

dowel pin   ´M2.5x14 inserted
locking with LOCTITE 648 coaxial

[image]
[image]

sgiraz, 2023-09-06:
Well done @fbiggi.

Closing as solved!

closed 2023-09-06 by sgiraz