robotology/mesh-support #1631: Jumps on the right ankle roll position mesaurements

GiulioRomualdi, 2023-09-06:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

Today while the robot was tethered to the crane, we heard the same noise of https://github.com/robotology/icub-tech-support/issues/1603. 

### Detailed context

We collected one dataset where the robot was supposed to be still and we noticed wired jumps on the right ankle roll position as can be seen in the following plot

[image]

Here is the dataset that can be used to deeply investigate the issue [robot_logger_device_2023_09_06_15_17_29.zip](https://github.com/robotology/icub-tech-support/files/12539736/robot_logger_device_2023_09_06_15_17_29.zip)

### Additional context

This may eventually cause the worsening of https://github.com/robotology/icub-tech-support/issues/1614

cc @S-Dafarra @lrapetti @mebbaid @isorrentino @paolo-viceconte

maggia80, 2023-09-07:
@AntonioConsilvio we could add the same ferrites used in the left ankle

sgiraz, 2023-09-07:
see also #1603 

cc @GiulioRomualdi

vigisushrutha23, 2023-09-10:
Today we also observed those jumps running some experiments on the robot

AntonioConsilvio, 2023-09-12:
I put the ferrite on the robot's ankle, so I would ask you to retest for the presence of these spikes.

[image]

cc @sgiraz @GiulioRomualdi

sgiraz, 2023-09-19:
Hi @GiulioRomualdi, @vigisushrutha23 

Do you have any feedback on this issue, after the fix applied by @AntonioConsilvio?

vigisushrutha23, 2023-09-19:
I head much lesser noise and the side-stepping seemed much smoother for my experiments done last week when compared to experiments done on or before 10-09. I shall add more feedback next week when I do more experiments.

sgiraz, 2023-09-25:
Hi @vigisushrutha23,   

Thanks for your feedback.

In the meantime, I will proceed to close this issue since it appears to have been resolved based on the feedback posted above. Please feel free to reopen it if needed.

cc @AntonioConsilvio

sgiraz, 2023-10-09:
Hi @vigisushrutha23, 
Did you have the opportunity to perform some tests as mentioned above?
If they went well, I would close this issue as the problem is solved.

sgiraz, 2023-10-18:
cc @GiulioRomualdi

GiulioRomualdi, 2023-10-19:
Hi @sgiraz in the last week, I didn't used the robot. However @paolo-viceconte @lrapetti and @evelyd used it quite a lot so they can comment on this

vigisushrutha23, 2023-10-19:
@sgiraz Sorry for the late reply. Simone and I had some some side-stepping experiments on the robot and after a few side-steps or in the second or third experiments, we observed jumps, irregular double stepping on the same foot and unstable walking. I am not entirely sure if they were only the right foot though because from surface level observation it seemed the jumps and double steps happened sometimes on the left leg as well.

lrapetti, 2023-10-19:
> However @paolo-viceconte @lrapetti and @evelyd used it quite a lot so they can comment on this

I did not observe particular behaviour.

paolo-viceconte, 2023-10-19:
> > However @paolo-viceconte @lrapetti and @evelyd used it quite a lot so they can comment on this
> 
> I did not observe particular behaviour.

Me neither. I also checked the logs of a few experiments we performed yesterday and they do not seem to contain spikes of the ankle rolls as those shown in https://github.com/robotology/icub-tech-support/issues/1631#issue-1884279938.

sgiraz, 2023-10-19:
> @sgiraz Sorry for the late reply. Simone and I had some some side-stepping experiments on the robot and after a few side-steps or in the second or third experiments, we observed jumps, irregular [...]

Hi @vigisushrutha23, 

Can you remember the time you experienced the issues. It is a quite important information for us in order to track the problem in relation to our maintenance activities. For example if the problem on the left ankle happen before the #1662 it may be already solved now.

vigisushrutha23, 2023-10-20:
@sgiraz I think we did the testing before the fix of #1662. We shall be testing today and let you know. If there are no problems this can be permanently closed .

vigisushrutha23, 2023-10-23:
@sgiraz  We didn't get to test the sidestepping due to some issue with parameters and lack of time. The normal walking worked just fine. So i think this issue can be closed.

sgiraz, 2023-10-23:
Ok, so I proceed to close this issue. 

Thanks to all for your feedback! 🦾

Feel free to reopen if necessary.

cc @AntonioConsilvio @vigisushrutha23 @lrapetti @GiulioRomualdi

closed 2023-10-23 by sgiraz