robotology/mesh-support #1632: `r_thumb_add` loose joint

SimoneMic, 2023-09-07:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The joint is loose and doesn't stay fixed in place. The motor is not actuated.

[image]

### Detailed context

[image]

### Additional context

Probably, during the various calibrations and turn-offs, the thumb collides with the upper-leg cover and it got damaged.

sgiraz, 2023-09-07:
Hi @SimoneMic, 

From the picture posted above It seems that a pin is coming out:

[image]

Additionally, please keep in mind that we are already aware of the necessity to replace the right forearm with a new one in order to solve many other problems.

@GiacomoVassallo13 could you take a look at it?

cc @fbiggi @AntonioConsilvio

fbiggi, 2023-09-08:
replacement of part code 16674 A1
gluing with Loctite 648 engine
MAXON_B7E7C938897D
replacement of the 2x6 pin on the support C_028_P_004 with screw m 2x8 + washer m2 + nut m2
movement test performed compliant
recovery time 2.5 hours

[image]
[image]

@AntonioConsilvio @sgiraz @maggia80 @Fabrizio69

maggia80, 2023-09-08:
the connector of the FAP of the thumb is damaged: 
[image]

this is due to the male connector that is damaged as well:

[image]

we should pay extreme attention to these cables/connectors during the insertion. If they are damaged we should replace them.

I guess the MMA of the right palm is damaged. We should verify it and replace it.

SimoneMic, 2023-09-11:
The adduction seems fine now, but there is another loose pin that doesn't keep the rest of the finger stiff:
[image]
[image]

Also I think that there's a missing spacer like this one (which was on the ground today):

[image]

EhsanRanjbari, 2023-09-12:
Ciao @AntonioConsilvio, What is the latest situation with the thumb? Is it possible to fix it anytime soon? (I reserved the robot today from 11 a.m. and I need the right hand to do some experiments), thanks

AntonioConsilvio, 2023-09-12:
Hi @EhsanRanjbari @SimoneMic, The phalange of the right thumb is broken:

[image]

This is how it should be

[image]

As you can see in this photo, the part of the phalanx where the hole was to connect the phalanx to the pivot is broken.
>
> [image]

So we should replace this phalange with a new one.

cc @maggia80 @sgiraz @Fabrizio69 @fiorisi

GiacomoVassallo13, 2023-09-13:
Hi. The broken component cod 16624.A3 hands thumb medial phalanx, hase been replaced. For this operation the finger was separated from the forearm.All the dowel pins were replaced and the new ones glued with Loctite648. 
The joint has been rewired due to disassembly of the thumb finger.
To rewire the joint it was necessary to separate the forearm from the robot, because the sheath had detached from the support from the side of the pulley.
The joint was tested with jogger set up before the arm was reattached.

cc: @Fabrizio69 @sgiraz @AntonioConsilvio

AntonioConsilvio, 2023-09-14:
Hi, After replacing the damaged phalanx by @GiacomoVassallo13, I solved the electronic problems (see [comment](https://github.com/robotology/icub-tech-support/issues/1632#issuecomment-1711787645)) by replacing the MMA board and some wires.

Following these interventions, only one problem remains, namely that the thumb_add manages to close (0-->90), but not to reopen (90-->0).

The problem seems to be caused by the mechanism which, following the interventions described in this [comment](https://github.com/robotology/icub-tech-support/issues/1632#issuecomment-1711273002), has become very hard. 

In fact it has been tested with the jogger and the mechanism works correctly, but with yarpmotorgui it does not.

cc @maggia80 @sgiraz @Fabrizio69

sgiraz, 2023-10-09:
Hi @SimoneMic,

Can you confirm that the joint involved in this issue is now working correctly?

If yes, I would close this issue.

SimoneMic, 2023-10-09:
Yep! The joint moves from the motorgui, and is not loose anymore.

Thanks