robotology/mesh-support #1634: Hardware fault in the right hand during the calibration

GiuliaScorzaAzzara, 2023-09-08:
### Robot Name 🤖

iCubGenova07 S/N:040

### Request/Failure description

The calibration is not completed correctly because of some joints going into a hardware fault in the right hand.
Moreover, the cables are exposed in the right middle finger because the proximal cable tie is missing.

### Detailed context

Hardware fault joints:
- r_thumb_oppose
- r_thumb_proximal

Not configured joints:
- r_thumb_distal
- r_index_distal
- r_middle_distal

AntonioConsilvio, 2023-09-19:
Hi @GiuliaScorzaAzzara, we went to erzelli recently for another task, and we took some time to better understand the nature of this problem. 
We came to the conclusion that the tendon of the thumb oppose of the right hand might be broken.

We apologise for not having intervened yet, but we are planning to intervene as soon as possible.

cc @sgiraz

GiuliaScorzaAzzara, 2023-09-19:
Hi @AntonioConsilvio,
Thank you for the update. When are you planning to come?
We have an important demo on Monday 25th, it would be great if you could fix it before that day.
Let me know. Thank you very much in advance!

sgiraz, 2023-09-25:
Hi @GiuliaScorzaAzzara,

I'm sorry for the late reply, unfortunately, we cannot satisfy your request in time for your demo. If you agree, we can plan the maintenance activity starting from the 2nd of October.

cc @AntonioConsilvio

GiuliaScorzaAzzara, 2023-09-27:
Hi @sgiraz,
The sooner the better for us.
Please let us know the day you plan to come within this week.
Thanks!

sgiraz, 2023-09-27:
Hi @GiuliaScorzaAzzara,

We are planning to have the intervention on the 2nd of October, is that fine with you?

cc @AntonioConsilvio @GiacomoVassallo13 @Fabrizio69

GiuliaScorzaAzzara, 2023-09-29:
Hi @sgiraz,
I'll be there. Thank you.
See you on Monday

AntonioConsilvio, 2023-10-03:
Hi @GiuliaScorzaAzzara, me and @GiacomoVassallo13 replaced the broken tendon (thumb oppose). 
Then we found some problems on the distal of the thumb, so we replaced the sheath and solved a wiring error on the motor connector, and now everything works properly.

cc @sgiraz @Fabrizio69

closed 2023-10-09 by sgiraz