robotology/mesh-support #1636: [Issue with network configuration]

ddetommaso, 2023-09-11:
### Robot Name 🤖

iCubGenova08 S/N:043

### Request/Failure description

The robot shows some connection issues probably due to some wrong network configuration.

- It takes a while after power on to be reachable at the usual ip (10.0.0.2)
- Sometime requests on ssh are refused (ssh: connect to host icub-head port 22: Connection refused)
- Sometime the robot is not reachable at the usual IP (10.0.0.2)

### Detailed context

Netplan configuration

`50-icub.yaml -> 50-icub-head-static.yaml.notload`

```
icub@icub-head:/etc/netplan$ cat 50-icub-head-static.yaml.notload

network:
  version: 2
  ethernets:
    enp4s0:
      dhcp4: no
      dhcp6: no
      addresses: [10.0.0.2/24]
      gateway4: 10.0.0.1
      nameservers:
        search: [icub.local]
        addresses: [10.0.0.1, 8.8.8.8]
      optional: true
    enp0s31f6:
      dhcp4: no
      dhcp6: no
      addresses: [10.0.1.104/24]
      optional: true
  wifis:
    wlp2s0:
      dhcp4: yes
      dhcp6: no 
      addresses: [10.0.0.3/24]
      access-points:
        "please-put-your-ESSID-here":
          password: "please-put-your-wifi-password-here"
      optional: true
```

davidelasagna, 2023-09-12:
Ciao @ddetommaso
 

> * It takes a while after power on to be reachable at the usual ip (10.0.0.2)

This may be caused since the wifi is not configured and it hangs trying to connect to an unexisting network until it goes in timeout, just comment the whole `wifis` section this should speed the things up.

> * Sometime requests on ssh are refused (ssh: connect to host icub-head port 22: Connection refused)

The ports can be closed If the system is not completely booted up eg:  the wifi issue mentioned before

  
> * Sometime the robot is not reachable at the usual IP (10.0.0.2)

This can be only because is not set to be static but the interface `enp4s0` it is on DHCP configuration are you sure that the symlink `50-icub.yaml` points `50-icub-head-static.yaml.notload` and not to `50-icub-head-dhcp.yaml.notload` ?
Can you post an `ls -la` of `/etc/netplan`  and a `cat /etc/netplan/50-icub.yaml` ?

Let Us know
Thanks!

@sgiraz @AntonioConsilvio

ddetommaso, 2023-09-12:
Hi @davidelasagna 

I can surely provide these information, but since the robot is located out of IIT I would need a couple of days to go there and do these checks. Could you please provide me also other logs that could be useful to report for having a better picture of the issue?

Thanks

davidelasagna, 2023-09-13:
@ddetommaso another output that can be useful is `ip a` launched on iCub-head, in this way I can also check the name of the interfaces.
Thanks!

ddetommaso, 2023-09-14:
Hi @davidelasagna 

the connection seems working having changed the netplan profile but still the issue sometime occurs. In the following some logs.

```
icub@icub-head:/etc/netplan$ ls -la
total 40
drwxr-xr-x   2 root root  4096 Sep 14 10:08 .
drwxr-xr-x 115 root root 12288 May 16  2022 ..
-rw-r--r--   1 root root   571 Jan 24  2019 50-icub-bonding-dhcp.yaml.notload
-rw-r--r--   1 root root   574 Sep 14 09:54 50-icub-bonding-static-nowifi.yaml.notload
-rw-r--r--   1 root root   781 Sep  8 16:51 50-icub-bonding-static.yaml.notload
-rw-r--r--   1 root root   352 Jan 24  2019 50-icub-dhpc.yaml.notload
-rw-r--r--   1 root root   345 Sep 14 08:14 50-icub-static-nowifi.notload
-rw-r--r--   1 root root   567 Jan 24  2019 50-icub-static.notload
lrwxrwxrwx   1 root root    35 Sep 14 10:08 50-icub.yaml -> 50-icub-bonding-static.yaml.notload
icub@icub-head:/etc/netplan$ cat 50-icub-bonding-static.yaml.notload 
network:
  version: 2
  ethernets:
    enp5s0:
      dhcp4: no
      dhcp6: no
      optional: true
    enp0s31f6:
      dhcp4: no
      dhcp6: no
      addresses: [10.0.1.104/24]
      optional: true
  wifis:
    wlp2s0:
      dhcp4: no
      dhcp6: no
      access-points:
        "please-put-your-SSSID-here":
          password: "please-put-your_Wifi-password-here"
      optional: true
  bonds:
    main-bond:
      addresses: [10.0.0.2/24]
      gateway4: 10.0.0.1
      nameservers:
        search: [icub.local]
        addresses: 
          - 10.0.0.1
          - 8.8.8.8
          - 8.8.4.4
      interfaces:
        - enp5s0
        - wlp2s0
      parameters:
        mode: active-backup
        primary: enp5s0
        mii-monitor-interval: 1
        gratuitious-arp: 5
```

```
icub@icub-head:/etc/netplan$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp0s31f6: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 00:e0:4b:65:43:ad brd ff:ff:ff:ff:ff:ff
3: enp5s0: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP> mtu 1500 qdisc fq_codel master main-bond state UP group default qlen 1000
    link/ether 12:c0:d9:34:c6:aa brd ff:ff:ff:ff:ff:ff
4: wlp2s0: <NO-CARRIER,BROADCAST,MULTICAST,SLAVE,UP> mtu 1500 qdisc mq master main-bond state DOWN group default qlen 1000
[... 31 lines cut ...]
icub@icub-head:/etc/netplan$ uname -a
Linux icub-head 4.15.0-177-lowlatency #186-Ubuntu SMP PREEMPT Thu Apr 14 20:56:23 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
icub@icub-head:/etc/netplan$ journalctl -p 4 -xb
-- Logs begin at Tue 2023-05-30 07:39:40 UTC, end at Thu 2023-09-14 11:29:02 UTC. --
Sep 14 10:10:02 icub-head kernel: ENERGY_PERF_BIAS: Set to 'normal', was 'performance'
Sep 14 10:10:02 icub-head kernel: ENERGY_PERF_BIAS: View and update with x86_energy_perf_policy(8)
Sep 14 10:10:02 icub-head kernel: MDS CPU bug present and SMT on, data leak possible. See https://www.kernel.org/doc/html/latest/admin-guide/hw-vuln/mds.html for more details.
Sep 14 10:10:02 icub-head kernel: TAA CPU bug present and SMT on, data leak possible. See https://www.kernel.org/doc/html/latest/admin-guide/hw-vuln/tsx_async_abort.html for more details.
Sep 14 10:10:02 icub-head kernel:  #3
Sep 14 10:10:02 icub-head kernel: (NULL device *): hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
Sep 14 10:10:02 icub-head kernel: Couldn't get size: 0x800000000000000e
Sep 14 10:10:02 icub-head kernel: MODSIGN: Couldn't get UEFI db list
Sep 14 10:10:02 icub-head kernel: Couldn't get size: 0x800000000000000e
Sep 14 10:10:02 icub-head kernel: Couldn't get size: 0x800000000000000e
Sep 14 10:10:02 icub-head kernel: Couldn't get size: 0x800000000000000e
Sep 14 10:10:02 icub-head kernel: sdhci-pci 0000:00:1e.6: failed to setup card detect gpio
Sep 14 10:10:05 icub-head kernel: (NULL device *): hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
Sep 14 10:10:05 icub-head kernel: (NULL device *): hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
Sep 14 10:10:05 icub-head kernel: thermal thermal_zone4: failed to read out thermal zone (-61)
Sep 14 10:10:05 icub-head systemd-udevd[467]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore 0' failed with exit code 99.
Sep 14 10:10:06 icub-head kernel: FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Sep 14 10:10:06 icub-head systemd-networkd[722]: Could not update link, ignoring: File exists
Sep 14 10:10:06 icub-head kernel: new mount options do not match the existing superblock, will be ignored

```

ddetommaso, 2023-09-14:
The network configuration seems to be solved configuring the icub-head with static address without wifi using the configuration as follows:

```
icub@icub-head:~$ ls -la /etc/netplan/50-icub.yaml 
lrwxrwxrwx 1 root root 22 Sep 14 12:40 /etc/netplan/50-icub.yaml -> 50-icub-static.notload
icub@icub-head:~$ cat /etc/netplan/50-icub-static.notload 
network:
  version: 2
  ethernets:
    enp5s0:
      dhcp4: no
      dhcp6: no
      addresses: [10.0.0.2/24]
      gateway4: 10.0.0.1
      nameservers:
        search: [icub.local]
        addresses: 
          - 10.0.0.1
          - 8.8.8.8
          - 8.8.4.4
      optional: true
    enp0s31f6:
      dhcp4: no
      dhcp6: no
      addresses: [10.0.1.104/24]
      optional: true

```

Anyway, often the head is not reachable with ssh even though the ip address is correct and I am actually able to ping. I am attaching here some additional logs.

[logicubgenova08.log](https://github.com/robotology/icub-tech-support/files/12609236/logicubgenova08.log)

cc @sgiraz

davidelasagna, 2023-09-14:
Analyzing the syslog, me and @sgiraz noticed some unusual reboots, around the same boot point
```
Sep 14 12:32:20 icub-head systemd[1]: Found device /dev/disk/by-uuid/ACB8-5BC2.
Sep 14 12:32:20 icub-head systemd[1]: Mounting /boot/efi...
Sep 14 12:32:22 icub-head systemd-networkd[567]: enp5s0: Gained carrier
Sep 14 12:32:22 icub-head kernel: e1000e: enp5s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
Sep 14 12:32:22 icub-head kernel: IPv6: ADDRCONF(NETDEV_CHANGE): enp5s0: link becomes ready
Sep 14 12:32:22 icub-head systemd-timesyncd[545]: Network configuration changed, trying to establish connection.
Sep 14 12:32:22 icub-head systemd-timesyncd[545]: Synchronized to time server 185.125.190.56:123 (ntp.ubuntu.com).
Sep 14 12:32:23 icub-head systemd-networkd[567]: enp5s0: DHCPv4 address 10.0.0.2/24 via 10.0.0.1
Sep 14 12:32:23 icub-head systemd-timesyncd[545]: Network configuration changed, trying to establish connection.
Sep 14 12:32:23 icub-head systemd-timesyncd[545]: Synchronized to time server 185.125.190.56:123 (ntp.ubuntu.com).
Sep 14 12:32:24 icub-head systemd-networkd[567]: enp5s0: Gained IPv6LL
Sep 14 12:32:24 icub-head systemd-networkd[567]: enp5s0: Configured
Sep 14 12:32:24 icub-head systemd-timesyncd[545]: Network configuration changed, trying to establish connection.
Sep 14 12:32:24 icub-head systemd-timesyncd[545]: Synchronized to time server 185.125.190.56:123 (ntp.ubuntu.com).
-- Reboot --
```
This behavior is usually due to a lack of power on the board, or an elettrical issue.
We recommend to check:
- power supply settings (Voltage and Current)
- the head power wires using a tester

sgiraz, 2023-10-18:
Hi @ddetommaso,

Have you experienced the problem anymore?
If not, I would close the issue.

ddetommaso, 2023-10-18:
Hi @sgiraz ,

today I was at the clinical center for maintanance and, yes I got the same issue. I think we should schedule an intervention once we finalized the data collection (around December). Since then we will need to move back the robot maybe it's worth to have firrst a deep check at CRIS. Let me know what you think it's the best.

sgiraz, 2023-12-04:
As agreed in a F2F chat with @ddetommaso, we are planning to transfer the robot only from Philos to CRIS by December 19th.

cc @Fabrizio69 @maggia80

sgiraz, 2023-12-12:
/remind 18th December maintenance intervention scheduled for tomorrow (cc @AntonioConsilvio @ddetommaso @Fabrizio69)

Fabrizio69, 2023-12-19:
Today, robot transferred to CRIS

cc: @sgiraz

sgiraz, 2024-01-09:
Hi @ddetommaso,

@davidelasagna and I tested the robot and after many tries, we have never experienced the problem mentioned in this issue.
As you can see from the video below, we can enter in `icub-head` over ssh after a few seconds (less than a minute) without any problems:

https://github.com/robotology/icub-tech-support/assets/18039045/ae455c9e-e06d-4eb4-8dd3-922505a3dea7

Anyway, we also tried to launch the `yarprobotinterface` and we spotted a problem **not mentioned in this issue**. In detail, the right-hand finger's adduction does not move/calibrate because the motor's pulley seems broken.

[image]

>[!Important]
>To resolve this issue, we need to extend the robot's downtime in CRIS beyond the originally planned schedule. If you don't require this joint to function correctly, we can remove its calibration from the configuration files and proceed with the shipment of the robot to Erzelli ASAP.

cc @Fabrizio69 @AntonioConsilvio @GiacomoVassallo13

ddetommaso, 2024-01-10:
Hi @sgiraz,

many thanks for the detailed report and the further tests.
If the network issue is related to the configuration of the power supply, then we need to schedule ad additional check on site to ensure that the issue won't happen again. Anyway if I am not wrong that power supply is not adjustable.

Regarding the other issue found, we would kind ask you to fix it because we need a fully functional robot since we are heavily using object grasping also acting on the abduction joint. 

Please consider that the starting of the training activities is approaching and any further delay respect to the originally planned schedule could be critical.

sgiraz, 2024-01-23:
Hi @ddetommaso,

Today we fixed the thumb's adduction. ✅
@AntonioConsilvio will post here a beautiful video showing that everything is working fine.

Then we'll proceed to deliver the robot in the week of 12th February as agreed on F2F chat

AntonioConsilvio, 2024-01-23:
Hi! here the video:

https://github.com/robotology/icub-tech-support/assets/114915464/c4b1ea76-71eb-4cac-9647-aa901b37288e

sgiraz, 2024-02-16:
The robot has been delivered successfully! 
One mounted on the Boggiano Pico's setup it works as expected!

If the unexpected restart will happen again I suggest to remove the UPS and connect the Power Supply directly to the wall socket.
In case the problem will persist after that, we'll have to investigate further, otherwise a new UPS would be needed.

As of now I consider this issue concluded successfully!