robotology/mesh-support #1637: [`ergoCubSN000`] not finishing calibration correctly

mebbaid, 2023-09-11:
### Robot Name 🤖

None

### Request/Failure description

When starting the robot, the robot does not finish the calibration sequence and throws the following error in the log, which I am not sure how to interpret.

```
	40,564850	ERROR			 from BOARD 10.0.1.22 (right_arm-eb22-j7_10), src LOCAL, adr 0, time 113s 762m 671u: (code 0x00000013, par16 0x0002 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. +
```

At the end, i tried to capture a video but the robot in the third time restarted correctly.
I am opening this issue in case the problem occurs again 

### Detailed context

The output of the log is the following one
[Uploading log_ergocub-torso_yarprobotinterface_2030.txt…]()

AntonioConsilvio, 2023-09-14:
Hi @mebbaid, I am quite sure that the problem was due this [issue](https://github.com/robotology/icub-tech-support/issues/1632#issuecomment-1711787645), where I fixed the electronical problems (see the [comment](https://github.com/robotology/icub-tech-support/issues/1632#issuecomment-1719057814)).

So give us feedback if the problem is still there or not.

cc @sgiraz

mebbaid, 2023-09-15:
The issue did not occur again.

sgiraz, 2023-09-15:
> https://github.com/robotology/icub-tech-support/issues/1637#issuecomment-1721025030

Nice! Closing.

closed 2023-09-15 by sgiraz