robotology/mesh-support #1638: Full overhaul of the robot

sgiraz, 2023-09-11:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

As per https://github.com/robotology/icub-tech-support/issues/1570#issuecomment-1633662274, we are required to do a full overhaul of the robot before delivering it back to Erzelli.

>**Note**
>This activity has been planned and managed internally.

### Detailed context

In order to perform the whole overhaul of the robot we have to perform the following tasks:
- [ ] Check the following tendons: shoulders, fingers, waist, and forearms
- [ ] Check for faulted or locked the motors 
- [ ] Check for faulted boards
- [ ] Check for faulted wires (CAN, ETH, SPI encoders)	
- [ ] Check for termination on mais boards and wherever they are needed
- [ ] Check for broken supports and covers
- [ ] Check for loosening and missing screws
- [ ] Check for possible shortcircuits (notified shutdowns [here](https://github.com/robotology/icub-tech-support/issues/1570) and [here](https://github.com/robotology/icub-tech-support/issues/1315))

### Additional context

Possible imporvements:
- Remove from the configuration files the following (unused) joints:
  - [ ] Whole lower body
  - [ ] Skin all over the robot

#### Robot tests + SW updates
- [ ] Update `robotology-superbuild`
- [ ] Update FW of all the boards
- [ ] `Yarprobotinterface` starts without errors
- [ ] Every single joint is movable correctly both by hand and by Yarpmotorgui
- [ ] Face expressions work
- [ ] Cameras work

AntonioConsilvio, 2023-09-12:
This is the PR for the configurations files (the robot is already up to date):

- https://github.com/robotology/robots-configuration/pull/565

The robot has been packed and it is ready for shipment:

[image]
[image]

cc @maggia80 @Fabrizio69 @sgiraz @ddetommaso @andreaderito @Andreamenini68

sgiraz, 2023-09-12:
see also: https://github.com/icub-tech-iit/tickets/issues/3084

closed 2023-09-12 by sgiraz