robotology/mesh-support #1639: One of the cables of the neck went out

GiulioRomualdi, 2023-09-14:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

During a walking experiment, one of the neck cables went out.

### Detailed context

Nothing

AntonioConsilvio, 2023-09-14:
Hi @GiulioRomualdi, the cable had remained intact, it had only come out of the cable tensioners. I have restored the windings and the neck should work properly now.

However, give us a feedback after your tests.

cc @sgiraz

mebbaid, 2023-09-14:
This happened again while walking with the normal walking controller.

[image]

mebbaid, 2023-09-14:
To go on with the experiments for our deadline we had to lock the neck using a fascette

GiacomoVassallo13, 2023-09-15:
Hi. The broken cable hase been replaced. Replaced also the screw of tensionators.

cc: @Fabrizio69 @sgiraz @AntonioConsilvio

sgiraz, 2023-09-15:
Super @GiacomoVassallo13!🚀

Closing!

closed 2023-09-15 by sgiraz