robotology/mesh-support #1643: [Give a meaningful title but don't enter the robot name here]

davidegorbani, 2023-09-21:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

icub-head cannot be pinged

### Detailed context

I was trying to turn on the robot, and when I tried to access `icub-head` (10.0.0.2) it was not possible to ping it; I was able to access the board 10.0.0.3 (dedicated to camera) and the output of `ip a` is:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 36:41:38:27:3b:db brd ff:ff:ff:ff:ff:ff
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 48:b0:2d:5f:d8:7f brd ff:ff:ff:ff:ff:ff
    inet 10.0.2.3/24 brd 10.0.2.255 scope global noprefixroute eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::4ab0:2dff:fe5f:d87f/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
4: enP5p1s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 00:0c:8b:b5:e4:b8 brd ff:ff:ff:ff:ff:ff
5: l4tbr0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether c2:ee:11:be:4d:89 brd ff:ff:ff:ff:ff:ff
6: rndis0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast master l4tbr0 state DOWN group default qlen 1000
    link/ether c2:ee:11:be:4d:89 brd ff:ff:ff:ff:ff:ff
7: usb0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast master l4tbr0 state DOWN group default qlen 1000
    link/ether c2:ee:11:be:4d:8b brd ff:ff:ff:ff:ff:ff
8: wlx1cbfced1ed2f: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 1c:bf:ce:d1:ed:2f brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.3/24 brd 10.0.0.255 scope global noprefixroute wlx1cbfced1ed2f
       valid_lft forever preferred_lft forever
    inet6 fe80::d5c8:332d:b5de:d2bf/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
9: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:b0:fd:1c:92 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
```

sgiraz, 2023-09-25:
Hi @davidegorbani 

Last Friday, @AntonioConsilvio, @davidelasagna and I checked the COMM express board mounted on the head. As a result after the offline tests, we can confirm that it has to be replaced with a brand-new one.

cc @Fabrizio69 @Miche19 @maggia80 @davidelasagna

maggia80, 2023-09-26:
we used to purchase the `SOM-7583C7-S8A1` from Mouser, but it's currently unavailable. I would suggest to go for the  `SOM-7583R7X-S8A1`, that should be identical but with extended temperature range.

[SOM-7583_DS_001.pdf](https://github.com/robotology/icub-tech-support/files/12729200/SOM-7583_DS_001.pdf)

gabrielenava, 2023-09-27:
thank you @maggia80, with @sgiraz we found a `COM-Express Type10, Advantech SOM i7-1185G7E 16GB DDR4, 64GB SSD` in the warehouse, I have opened a ticket for it and now we are configuring it to install Ubuntu 22.04

maggia80, 2023-09-27:
yes @gabrielenava please open a purchase issue for replacing the one you took from the warehouse. It was the last one.

gabrielenava, 2023-09-28:
Today we tried to install the OS on the COM-Express board. There were several issues to overcome.  In the end, we just managed to power on the board and connect it to a monitor, and enter in the BIOS. From there, it was not possible to do anything as the keyboard or mouse were not connecting. We tried with several different keyboards and none of them worked.

The task seems much more complex then originally expected, and we don't have the right equipment to work on it (for example, we were not able to connect the board to any monitor available in the lab until @sgiraz lent us a monitor from icub-tech that worked): at this point, the safest option for us would be to wait when icub-tech will be available to carry on the task.

cc @sgiraz @DanielePucci @maggia80 @davidegorbani

davidelasagna, 2023-10-03:
Yesterday, along with @gabrielenava and @davidegorbani, I've successfully installed and configured the OS on the COM-express of icub-head.
Today, there has been an issue on the network side due to a faulty configuration regarding the wifi interface, but it has already been fixed.

cc @maggia80 @sgiraz @DanielePucci

sgiraz, 2023-10-09:
Thanks for your precious support @davidelasagna!

Closing as solved.

closed 2023-10-09 by sgiraz