robotology/mesh-support #1644: Torso yaw not controllable

S-Dafarra, 2023-09-22:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

During the IIT 20y demo, the torso yaw suddenly jumped to the limit. Checking the motorgui, we noticed that in zero position the joint encoder was measuring more than 36deg. We rebooted and then the joint position was fine, but it was not possible to control the joint anymore, and the robot was going into a hardware fault complaining about a missing index in the encoder. 

Unfortunately, we do not have the logger because it was saturated.

### Detailed context

There seem to be two issues:
- The LCORE seems not working
- The AMO is subject to jumps (as in https://github.com/robotology/icub-tech-support/issues/1602#issuecomment-1649599334). We would need the incremental calibration also for this joint I think

### Additional context

During the demo @maggia80 had to short circuit the phases of the motor in order to have it stiff enough during the demo.

maggia80, 2023-09-22:
cc: @ale-git , @valegagge

ale-git, 2023-09-27:
About the yaw joint getting stuck in one position, as I said yesterday to @AntonioConsilvio I think that it is a motor encoder issue. Even if it seems to work properly with the jogger board, maybe in some position the wire is stressed and electrical contact fails. I suggest for the fist thing to change the cable, and if it doesn't work to change the motor encoder sensor as well. 

About the AMO sensor the problem is probably the same we had on almost all the other AMO sensors, so the solution is likely switching to hard stop calibration. We have to program a proper calibration sequence in order to reduce the stress on motors and thus prevent I^2t possible issues.

AntonioConsilvio, 2023-09-29:
Hi, me and @Gandoo carried out several interventions, including cleaning the optical disc, replacing LCORE5 and the cable connecting LCORE5 to 2FOC.

We also put ferrites on the three motor phases, the LCORE cable and the AMO sensor: 

[image]

We tested the joint for a long time, in different configurations and speeds, and the problem did not occur again.

However, I would keep the issue open for now, in case the problem returns with the next uses of the robot.

cc @maggia80 @sgiraz @Fabrizio69 @S-Dafarra @lrapetti @ale-git

S-Dafarra, 2023-09-29:
Great @AntonioConsilvio!

I would consider what @ale-git wrote too

> About the AMO sensor the problem is probably the same we had on almost all the other AMO sensors, so the solution is likely switching to hard stop calibration. We have to program a proper calibration sequence in order to reduce the stress on motors and thus prevent I^2t possible issues.

I think that it is still feasible to apply the incremental calibration on the torso yaw. This should make this problem even less likely to occur again.

AntonioConsilvio, 2023-10-11:
Hi @S-Dafarra, @ale-git did incremental calibration for the torso joints, here is the PR:

- https://github.com/robotology/robots-configuration/pull/577

As two weeks have passed without any negative feedback, I proceed to close the issue. If the problem reoccur feel free to reopen this issue.

cc @maggia80 @sgiraz @lrapetti

closed 2023-10-11 by AntonioConsilvio