robotology/mesh-support #1646: The left wrist starts in hardware fault and cannot be controlled

S-Dafarra, 2023-09-22:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

In the end of the IIT20y demo, the left wrist went into hardware fault and it was not possible to recover it.

### Detailed context

It happened all of a sudden, the robot was being controlled, but no one was touching it. It went into hardware fault and neither by restarting it it was possible to restore it

AntonioConsilvio, 2023-09-26:
Hi @S-Dafarra, see the comment:

- https://github.com/robotology/icub-tech-support/issues/1642#issuecomment-1735011980

sgiraz, 2023-10-09:
Closing as per https://github.com/robotology/icub-tech-support/issues/1642#issuecomment-1752611984

closed 2023-10-09 by sgiraz