robotology/mesh-support #1647: Left arm wrist does not calibrate

PasMarra, 2023-09-25:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

Today I tried to start up the robot, but the left wrist did not calibrate.
After opening the yarpmotorgui I realized that the encoder of the wrist are not working or seems to be. Moreover, the shoulder and other parts are not calibrating as well.

[image]

### Detailed context

I have attached the log from the yarprobotinterface.
[log_icub-head_yarprobotinterface_1775.txt](https://github.com/robotology/icub-tech-support/files/12717455/log_icub-head_yarprobotinterface_1775.txt)

sgiraz, 2023-10-18:
Hi PasMarra, the maintenance is scheduled for Wednesday 25th of October.

cc @GiacomoVassallo13 @AntonioAzocar @Fabrizio69

sgiraz, 2023-10-18:
/remind 24 October Maintenance activity scheduled for tomorrow

cc @GiacomoVassallo13 @AntonioAzocar @Fabrizio69

xEnVrE, 2023-10-20:
> 24 October Maintenance activity scheduled for tomorrow

Is the maintenance scheduled for October 25 then? Is not the robot involved in the demos for the PI16?

Thank you

pattacini, 2023-10-20:
>Is not the robot involved in the demos for the PI16?

Yes!

cc @martinaxgloria

martinaxgloria, 2023-10-20:
@xEnVrE the demos for the PI16 are on October 25th during the afternoon and iCubGenova11 is involved. 

@sgiraz could you move the maintenance activity to the day after, please?

AntonioAzocar, 2023-10-30:
Hi @xEnVrE, I found the black cable of the AEA2 (**Name: AE6 2S2**) broken, so I re-soldered it.      
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          

[image]

The **left arm wrist** is working properly again, but we've found other problems.
As you can see in this photo, the thumb oppose has a problem to encoder that reads very high values.
We should do more test to understand better the problem.

[image]

Thanks @martinaxgloria for your help!

cc @sgiraz @AntonioConsilvio

xEnVrE, 2023-10-30:
Thanks for the update @AntonioAzocar.

Do you want me to open a dedicated issue in order to fix that so that we can close this one - as the wrist is now working as expected?

Thanks

AntonioAzocar, 2023-10-30:
Yes, perhaps it would be better to have a dedicated issue, thanks!

cc @sgiraz

sgiraz, 2023-10-30:
>The left arm wrist is working properly again, but we've found other problems.
As you can see in this photo, the thumb oppose has a problem to encoder that reads very high values.
We should do more test to understand better the problem.

Hi @xEnVrE feel free to open a new issue in order to track the problemi mentioned by @AntonioAzocar in the comment above.
I proceed to close this issue.

closed 2023-10-30 by sgiraz