robotology/mesh-support #1649: Inconsistency in the orientation data between head IMU (assumed to be correct) and feet IMUs (possibly wrong)

G-Cervettini, 2023-10-04:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

Misalignment between the expected IMU rotation and measured one, in a similar way as shown in the issue https://github.com/icub-tech-iit/task-force-miscellanea/issues/126.

### Detailed context

Yesterday with @GiulioRomualdi and @DanielePucci we tested my imu-based [floating base estimator](https://github.com/G-Cervettini/bipedal-locomotion-framework/blob/fix_imu_base_estimation/src/Estimators/src/BaseEstimatorFromFootIMU.cpp) on the real robot for the first time and the data stream from right foot IMUs showed an unexpected behavior.  
The estimator is supposed to track the robot pose changes based on IMU data and it is working in [simulation](https://github.com/ami-iit/element_floatingbase-exteroceptive-estimation/issues/25).  
Here the behavior of the estimator in yesterday's test on the robot  

***NO CORRECTION:***

https://github.com/robotology/icub-tech-support/assets/101020439/6162ff23-e032-4732-ad7e-15adf0d387ed

Having a look  at this result, it looked like Roll and Pitch were switched wile Yaw had the wrong sign. Thus, I tried to apply [this](https://github.com/icub-tech-iit/task-force-miscellanea/issues/126#issuecomment-1185583474) correction to imu data and the result matched much better the expected behavior  

***CORRECTED:***

https://github.com/robotology/icub-tech-support/assets/101020439/1674dd3a-a5eb-4fe4-ba3e-be5af6dcbb36

Thus, I'm involving icub-tech because there may be something to fix in the F/T firmware regarding RPY data.  
In my estimator I'm using both the right foot IMUs data and applying the `iDynTree::geodesicL2MeanRotation()` on them.  

With @GiulioRomualdi we also tried to consider the front and rear sensors separately and they seem to behave differently from each other:  

https://github.com/robotology/icub-tech-support/assets/101020439/7d338a15-9933-4e1f-99dd-a84ba2f8ed50

In the previous video the root link imu data are used as ground-truth and the FK-propagated right foot orientation (thick axes) is compared with raw data from front and rear foot IMUs (thin axes).  
In particular the front imu seems to show a much bigger error than the rear one.
Thanks in advance for the support!

### Additional context

Here the mat files of the logged data:  
[03-Oct-2023_Feet_IMUs_Base_Estimation.zip](https://github.com/robotology/icub-tech-support/files/12804645/03-Oct-2023_Feet_IMUs_Base_Estimation.zip)

### How does it affect you?

I am working on a imu-based floating base estimation algorithm and I need IMUs data to be consistent for my code to behave as expected.

DanielePucci, 2023-10-05:
CC @Nicogene @maggia80

Nicogene, 2023-10-05:
Hi @G-Cervettini,

analyzing this issue

- https://github.com/robotology/icub-main/issues/901

We discovered that when we fixed the measurement frame for the accelerometer gyro etc. changing the placement option of the chip, we forgot to consider that the euler representation we use in [YARP](https://www.yarp.it/latest/classyarp_1_1dev_1_1IOrientationSensors.html#adb2a40d03c747aa9c8b7fc542f193fd9) is different from any orientation definition that imu bosch chip supports ([windows or android standard](https://github.com/robotology/icub-main/issues/901#issuecomment-1731627499)).

For this reason, we need to check both the `mtb4`, `strain2`, and `rfe` and apply for each of them a transformation that allows us to correct the Euler angles.

The fix for the rfe should be already done by @davidetome, as soon as we receive positive feedback we will move to the other boards.

cc @pattacini @marcoaccame

davidetome, 2023-10-11:
Ciao @G-Cervettini !

 after [this fix](https://github.com/robotology/icub-main/issues/901#issuecomment-1757325799), we also fixed the `STRAIN2`/`STRAIN2C`  and tested on the desk; can you please flash these binaries in the `RFE `and in the `STRAIN2 `(or `STRAIN2C`) and test again?

[binaries.zip](https://github.com/robotology/icub-tech-support/files/12868141/binaries.zip)

We are waiting for your kind positive feedback to merge this [PR](https://github.com/robotology/icub-firmware/pull/425)

cc @Nicogene @marcoaccame @sgiraz

G-Cervettini, 2023-10-11:
Ciao @davidetome, thanks for your fix! I will test again and update you asap.  
Once the [PR](https://github.com/robotology/icub-firmware/pull/425) is merged both `rfe` and `strain2` should be ok. What about `mtb4` instead, as @Nicogene was mentioning?  

> For this reason, we need to check both the `mtb4`, `strain2`, and `rfe` and apply for each of them a transformation that allows us to correct the Euler angles.

davidetome, 2023-10-11:
> Ciao @davidetome, thanks for your fix! I will test again and update you asap. Once the [PR](https://github.com/robotology/icub-firmware/pull/425) is merged both `rfe` and `strain2` should be ok. [...]

`MTB4 `has the same chip placement as the RFE and then has the same Euler angles conversion.

davidetome, 2023-10-11:
> We are waiting for your kind positive feedback to merge this https://github.com/robotology/icub-firmware/pull/425

The robot got broken, we've to wait for the fix 

https://teams.microsoft.com/l/message/19:def07f83a09d443eb576b6ddf10101af@thread.tacv2/1697025822546?tenantId=bf2f465e-1f75-4b6f-be29-bf53273bc1da&groupId=0ad0a501-ae56-494f-8425-9f5d08c9763f&parentMessageId=1697025822546&teamName=iCub-CRIS&channelName=ergoCub%20-%20SN000&createdTime=1697025822546&allowXTenantAccess=false

sgiraz, 2023-10-12:
Hi @davidetome, afik the robot has been fixed.

cc @G-Cervettini @Nicogene

GiulioRomualdi, 2023-10-12:
One question before flashing. Which  version of yarp, icub-main and icub-firmware-shared is required to have the firmware up and running?

davidetome, 2023-10-13:
> One question before flashing. Which version of yarp, icub-main and icub-firmware-shared is required to have the firmware up and running?

Ciao @GiulioRomualdi , sorry for late reply, I missed your comment,  th FW modified is the last in `icub-firmware` devel

GiulioRomualdi, 2023-10-13:
Ciao @davidetome, this morning we tested the firmware on the strain2 as follows:
1. we moved the robot on the pole
2. I moved manually the left leg as in the following video 

   https://github.com/robotology/icub-tech-support/assets/16744101/6aab6816-c26d-4d8e-8804-40f266dd1c50
3. We collected the data of the IMUs on the strains and we compared them with the expected values of the forward kinematics by using https://github.com/ami-iit/xcub-sensors-check/tree/main

We flashed the firmware you gave us in https://github.com/robotology/icub-tech-support/issues/1649#issuecomment-1757334894 and we repeated the experiment again

## Original firmware strain2 v2.2.0
|  L_FOOT_FRONT_ORIENTATION | L_FOOT_REAR_ORIENTATION |
|:-------------------------:|:-----------------------:|
| [image] | [image] |

## New firmware strain2 v2.3.0

|  L_FOOT_FRONT_ORIENTATION | L_FOOT_REAR_ORIENTATION |
|:-------------------------:|:-----------------------:|
| [image] |  [image] |

So now I think there is a problem in the pitch

You can find here the associated datasets: [strain_imu_test.zip](https://github.com/robotology/icub-tech-support/files/12893342/strain_imu_test.zip)

Nicogene, 2023-10-13:
> So now I think there is a problem in the pitch

In fact @davidetome and I had some doubt about the pitch sign that was the only one different from the rfe. This means that we can get rid of the #ifdef!

davidetome, 2023-10-13:
fix done , see : 

https://github.com/robotology/icub-main/issues/901#issuecomment-1761702534

pattacini, 2023-10-13:
I would keep it open until the final user test is passed.

davidetome, 2023-10-16:
Ciao @GiulioRomualdi!  New binaries are available in `devel` branch after this merge : 

- https://github.com/robotology/icub-firmware-build/pull/108#event-10661108824

Could you please test them as suggested by @pattacini? Then we can close the issue 👍🏻 

cc @Nicogene @marcoaccame

GiulioRomualdi, 2023-10-16:
Hi @davidetome, thank you for the effort. I will try to test it this week if I find a free slot on ergoCub. if this is not possible, a possibility is to perform the test on ergoCubSN001 as soon as it is available

GiulioRomualdi, 2023-10-16:
Hi all! I found a free slot on the robot and I tested the new firmware taken from https://github.com/robotology/icub-firmware-build/commit/a910398e2afd18b430cb23ba669f5895813b2977

|  L_FOOT_FRONT_ORIENTATION | L_FOOT_REAR_ORIENTATION |
|:-------------------------:|:-----------------------:|
|  [image] |  [image] |

Everything is working as expected 🎉 Thank you all for the help! 

⚠️ I reverted the old firmware (v2.2.0) to comply with the Distro release.

pattacini, 2023-10-17:
That's great!
Thanks for testing.

closed 2023-10-17 by pattacini