robotology/mesh-support #1650: Right knee electrical cable disconnected from the connector

gabrielenava, 2023-10-09:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

An electrical cable associated with the right knee disconnected from its connector during robot startup. I presume it happened after multiple start and stop of the robot while we were debugging another issue.

cc @davidegorbani @DanielePucci 

### Detailed context

[image]

gabrielenava, 2023-10-10:
thank you @AntonioAzocar!

sgiraz, 2023-10-10:
Well done @AntonioAzocar! 
Closing as solved.

closed 2023-10-10 by sgiraz