robotology/mesh-support #1651: Neck Tendon Snapped

SimoneMic, 2023-10-10:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The neck tendon is broken as per picture:
[image]

### Detailed context

The cable snapped suddenly after starting the walking controller. The robot collapsed on itself and the neck broke.
There weren't any error messages on the walking-controller side, meanwhile the motorgui showed hardware fault for the head.

### How does it affect you?

Decreases the quality of the sensor readings on the head, reducing the localization and navigation capabilities.

GiacomoVassallo13, 2023-10-11:
Hi. this morning @fbiggi and me have replaced the broken tendon. Actualy the robot have't got the che face cover, because the cable of microphone is damneged. @Gandoo is doing the repairs.

cc: @sgiraz @Fabrizio69

GiacomoVassallo13, 2023-10-12:
HI. Yesterday, because a damaged component, we had to separate the head from the robot to work on it on the desk. It was the omega-shaped neck component that became deformed. The damaged component has been replaced with a new one. 
Moved mechanically the joint moves without obstructions or anomalies of any kind.
I await feedback on the movement under control.

cc: @AntonioConsilvio @sgiraz

closed 2023-10-13 by sgiraz