robotology/mesh-support #1652: Right hip roll in hardware fault at startup

gabrielenava, 2023-10-12:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The right hip roll goes in hardware fault at startup. The joint is not recoverable even later on from motorgui, pressing idle or calib. the encoder measurement is strange (110 degrees instead of zero).

I will add pictures and logger while we try again the startup phase for debugging the problem.

### Detailed context

cc @davidegorbani @vpunithreddy

davidegorbani, 2023-10-12:
Here is the picture of the `yarpmotorgui` related to the `r_hip_roll`, the file of the `yarplogger` when we started the robot and a picture of the cable of the right leg.

[image]

[image]

[right_leg_overcurrent.zip](https://github.com/robotology/icub-tech-support/files/12879947/right_leg_overcurrent.zip)

DanielePucci, 2023-10-17:
Probably it would be better to add the glue when fixing the connector @AntonioConsilvio @sgiraz

AntonioConsilvio, 2023-11-10:
Hi @gabrielenava, the problem was caused by failure to read the motor encoder (LCORE) connected to 2FOC 0B10.

The sensor was not reading because the connector on the 2FOC board stopped working.

So I replaced the 2FOC with a new one, and the robot was tested and it works properly now!

cc @sgiraz

gabrielenava, 2023-11-10:
>So I replaced the 2FOC with a new one, and the robot was tested and it works properly now!

thank you @AntonioConsilvio, but I would wait before closing this issue: I have heard that @HosameldinMohamed tested the robot yesterday and the jip roll was going in hardware fault. when moved. 

Waiting for @HosameldinMohamed to have more details and confirm that there is still this problem!

gabrielenava, 2023-11-10:
EDIT: I see that @HosameldinMohamed just opened https://github.com/robotology/icub-tech-support/issues/1679 for tracking this problem. Being the same joint, I think the problem is related to the one of this issue, but since a new issue has been opened we can close this one and continue the analysis in the other issue.

HosameldinMohamed, 2023-11-10:
Oh sorry I just saw this issue!

sgiraz, 2023-11-10:
> Oh sorry I just saw this issue!

just for reference: https://github.com/robotology/icub-tech-support/issues/1679

davidegorbani, 2023-11-10:
Here is the plot of the current of the right hip roll when moving it of about 20 degrees.

[image]

GiacomoVassallo13, 2023-11-14:
Hi. The faulted motor was the hip roll of right leg not the waist. 
The faulted motor has been replace wich a new one. 
I attach the photo of the replaced stator:

[image]

The optical disc was also replaced as it was worn ,and the correct reading was verified with the bench test.

Unfortunately, I would like to point out that the assembly is missing some components: in particular the front edges of the sprockets and a  piece of the belt tensioner.

[image]

This components will be added the as they become available.

The motor group was reassembled on the robot. @AntonioAzocar will take care of the electronic wiring of the new engine.

I attend the feedback. 

cc: @Fabrizio69 @sgiraz @AntonioConsilvio

HosameldinMohamed, 2023-11-15:
Today @AntonioConsilvio checked the AMO sensor using https://icub-tech-iit.github.io/procedures/tp-amo-calibration/#magnetic-ring-or-amo-board-incorrectly-mounted (you may add details if you prefer), and it seemed OK!

Then he performed a stator-rotor calibration https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub3-stator-rotor-autocalibration/

Now the joint is working as expected. We moved the repaired joint (hip roll) and it moves without a fault. Also, comparing with the left one, the current values seem similar, However, the estimated torque values are different!

AntonioConsilvio, 2023-11-15:
Hi! Thanks @HosameldinMohamed for the summary! 

Regarding the AMO, what I did was to perform the calibration without installing on the RAM the results. I did this just to see if the magnetic ring was mounted correctly (see an example in the [documentation](https://icub-tech-iit.github.io/procedures/tp-amo-calibration/#magnetic-ring-or-amo-board-incorrectly-mounted)).

The robot is working correctly now!

> [!Note]
> I would also like to report an anomaly, in fact during the rotor-stator calibration, I saw that the old offset value present in the hip roll joint was 0 (inside the mechanicals file). 
That offset comes from the rotor-stator calibration, and as far as I know, it can't be 0, so its presence in the configuration files is strange.
> 
> Maybe that negatively affected the motor and caused it to burn out.

However, since the problem is solved, I'll close to the issue!

cc @sgiraz @gabrielenava

closed 2023-11-15 by AntonioConsilvio